kroshu/kuka_drivers #10: Why not just create a hardware interface for ros2_control?

resizoltan, 2022-03-27:
The de-facto control framework for ROS is ros_control, which is currently being ported to ROS2. A detailed reasoning should be provided why nodes are created for the driver instead of just implementing a hardware interface so that the robot can be used with ros2_control.

kovacsge11, 2022-04-04:
The main reason was that the porting of ros_control was still undergoing, right? Or am I missing something? Are there any further reasons? @resizoltan

kovacsge11, 2022-04-05:
As @Svastits correctly stated, he added some extra features already which are useful for his project. We don't see yet how exactly these features could be mapped to ros2_control, we're not entirely sure that it's possible. We'll take a look at it.

Groebehavn, 2022-04-11:
Could you name the features you added, which might be problematic? @Svastits @kovacsge11 ?

resizoltan, 2022-04-14:
These types of features are very useful and a core element of any control stack imo. I don't know how exactly they are implemented in ros2_control. But I don't think this issue has any influence on our work right now, since this does not invalidate what we want to do, only shows a desired control stack functionality that has to be implemented if this approach is pursued.

closed 2022-04-24 by resizoltan