kroshu/kuka_drivers #9: Add minimal documentation

resizoltan, 2022-03-27:
There is currently no public documentation on the usage of the driver. Add minimal documentation describing the architecture, system setup and a quick-start guide.

Proposal: create a doc folder with Markdown files that are then exported to the repository Wiki using the GitHub action [wiki-page-creator-action](https://github.com/Decathlon/wiki-page-creator-action)

Groebehavn, 2022-03-31:
ROS Wiki states, that: "The ROS Wiki is for ROS 1." There is a "rolling documentation" for ROS2. Probably uploading to the Wiki might not be necessary. What's your opinion?

resizoltan, 2022-03-31:
I mean the Wiki page on GitHub: https://github.com/kroshu/ros2_kuka_sunrise/wiki
I think this is a good compromise between having just some markdown files (which is a bit more hassle to navigate for users) and a separate documentation webpage (which is more work for us to set up, but in the future it might be the best way to go - see for example https://control.ros.org/getting_started.html )

resizoltan, 2022-03-31:
So the idea is we just write the documentation in our favourite desktop md editor and can track it via git but also make it visible on the Wiki page where navigation is somewhat better

Groebehavn, 2022-03-31:
Okay, misunderstood. Sounds good! :)

resizoltan, 2022-04-08:
Hey, I pushed an initial documentation to [feature/prepare-for-open-collab](https://github.com/kroshu/ros2_kuka_sunrise/tree/feature/prepare-for-open-collab). I guess it should be enough for now.
@kovacsge11 it would be nice if you could double-check the [quick-start-guide](https://github.com/kroshu/ros2_kuka_sunrise/blob/feature/prepare-for-open-collab/doc/1.-Quick-start-guide.md)
@Groebehavn if you have time sometime, you could take a look at the [ros2_control related section](https://github.com/kroshu/ros2_kuka_sunrise/blob/feature/prepare-for-open-collab/doc/3.-Relationship-to-ros2_control.md) and we could discuss whether you agree

closed 2022-04-24 by resizoltan