kroshu/kuka_drivers #14: Evaluate GitHub actions for pipeline builds

resizoltan, 2022-04-08:
Industrial CI also runs on GitHub actions. Main reason to consider switching: Travis-CI does not support skipping builds if only certain files or directories have changed (travis-ci/travis-ci#6301). This would be quite relevant for the documentation folder - it makes no sense to trigger a build when updating it.

kovacsge11, 2022-04-11:
I fully support this. I've read the same about skipping builds.

Groebehavn, 2022-04-11:
PR into master, great! :) Can we consider this as closed? All builds are migrated? Or are there any left out?

kovacsge11, 2022-04-11:
It's only done for this repo, and it's still sitting on prepare-for-open-collab branch. When we have an initial somewhat-ready state, we can merge to master imo.

closed 2022-04-13 by resizoltan