kroshu/kuka_drivers #61: KSS RSI driver pimp up with robot manager node

altex111, 2023-02-24:


Svastits, 2023-06-19:
Heap allocations in control loop should also be removed!
RSICommand heap-allocates in consctructor, and it is instantiated in every cycle (write)

altex111, 2023-06-22:
Tasks:
- [ ] Heap allocation in udp server
- [ ] Heap allocation in control loop and RSI Command
- [ ] Redesign RSI Command and RSI State so it makes direct connection between buffer and hw param
- [ ] working stop action when ROS side stopped
- [ ] working stop action when robot side stopped
Low prio tasks:
- [ ] IP set via config file
- [ ] add cartesi control
- [ ] controller handler and control mode change integration

altex111, 2023-09-26:
## Evaluating PUGI XML

After creating my own xml parser i implemented the [pugi ](https://pugixml.org/) xml parser too. The parser is capable of working on C style character strings which is compatible with the udp library i use. The user is also able to redefine the allocation functions. I redefined them with the [memory resources](https://en.cppreference.com/w/cpp/header/memory_resource) library so the memory is allocated at the start of the algorithm.

### Results
As the following diagrams shows there is a significant difference between the two xml parser both in encodeing and decodeing.
#### Decode
[image]
#### Encode
[image]

### Conclusion and Tasks
The driver for encodeing and decoding messages to and from the RSI should use the PUGI XML parser. However further tests should be created to prove the real-time nature. The PUGI can be configured multiple ways.