kroshu/kuka_drivers #75: [KukaRoXHardwareInterface]: SetQoSProfile failed

jlee4176901, 2023-03-09:
**Relevant Info:**
- OS: Ubuntu 22.04 with real-time patch.
- ROS version: Humble (default DDS).
- ros2_kuka_drivers branch: 0.6.0.

**Issue:**
Trying to follow readme [https://github.com/kroshu/ros2_kuka_drivers/wiki](url). The error appears in the life cycle management, when executing the command:

`ros2 lifecycle set robot_manager configure`

No hardware is involved. Appears for both torque and position interface.

**Log:**
```
1678376094.9377315 [INFO] [launch]: All log files can be found below /home/ros2/.ros/log/2023-03-09-16-34-54-937012-ros2-Precision-Tower-5810-20489
1678376094.9379082 [INFO] [launch]: Default logging verbosity is set to INFO
1678376095.3964398 [INFO] [rox_control_node-1]: process started with pid [20503]
1678376095.3967752 [INFO] [robot_manager_node-2]: process started with pid [20505]
1678376095.3969758 [INFO] [robot_state_publisher-3]: process started with pid [20507]
1678376095.3971589 [INFO] [spawner-4]: process started with pid [20509]
1678376095.3973446 [INFO] [spawner-5]: process started with pid [20511]
1678376095.3975382 [INFO] [spawner-6]: process started with pid [20513]
1678376095.3977170 [INFO] [spawner-7]: process started with pid [20515]
1678376095.4304569 [rox_control_node-1] [INFO] [1678376095.430076332] [resource_manager]: Loading hardware 'iisy_hardware' 
1678376095.4364066 [robot_state_publisher-3] [INFO] [1678376095.435909475] [robot_state_publisher]: got segment LBR3R760_iisy_axis1
1678376095.4366567 [robot_state_publisher-3] [INFO] [1678376095.436030796] [robot_state_publisher]: got segment LBR3R760_iisy_axis2
1678376095.4367929 [robot_state_publisher-3] [INFO] [1678376095.436048584] [robot_state_publisher]: got segment LBR3R760_iisy_axis3
1678376095.4369311 [robot_state_publisher-3] [INFO] [1678376095.436057980] [robot_state_publisher]: got segment LBR3R760_iisy_axis4
1678376095.4370553 [robot_state_publisher-3] [INFO] [1678376095.436066488] [robot_state_publisher]: got segment LBR3R760_iisy_axis5
1678376095.4371786 [robot_state_publisher-3] [INFO] [1678376095.436074846] [robot_state_publisher]: got segment LBR3R760_iisy_axis6
1678376095.4373093 [robot_state_publisher-3] [INFO] [1678376095.436083197] [robot_state_publisher]: got segment LBR3R760_iisy_fixedbase
1678376095.4374912 [robot_state_publisher-3] [INFO] [1678376095.436091720] [robot_state_publisher]: got segment LBR3R760_iisy_flange
1678376095.4377170 [robot_state_publisher-3] [INFO] [1678376095.436100070] [robot_state_publisher]: got segment base_link
1678376095.4441547 [robot_manager_node-2] [INFO] [1678376095.443825865] [robot_manager]: Starting Robot Manager Node init
1678376095.4446981 [rox_control_node-1] [INFO] [1678376095.444162690] [resource_manager]: Initialize hardware 'iisy_hardware' 
1678376095.4454660 [rox_control_node-1] [INFO] [1678376095.445255036] [KukaRoXHardwareInterface]: Init successful
1678376095.4456372 [rox_control_node-1] [INFO] [1678376095.445277722] [resource_manager]: Successful initialization of hardware 'iisy_hardware'
1678376095.4460344 [rox_control_node-1] [INFO] [1678376095.445288258] [KukaRoXHardwareInterface]: Export state interfaces
1678376095.4463315 [rox_control_node-1] [INFO] [1678376095.445393494] [KukaRoXHardwareInterface]: Export command interfaces
1678376095.4465237 [rox_control_node-1] [INFO] [1678376095.445544403] [resource_manager]: Hardware Component with name '' does not exists
1678376095.4470041 [robot_manager_node-2] [INFO] [1678376095.446758321] [robot_manager]: Control mode changed to POSITION_CONTROL
1678376095.4473732 [robot_manager_node-2] [INFO] [1678376095.446975492] [robot_manager]: IP address of controller: 0.0.0.0
1678376095.7861826 [rox_control_node-1] [INFO] [1678376095.784359823] [controller_manager]: Loading controller 'joint_impedance_controller'
1678376095.7997952 [rox_control_node-1] [INFO] [1678376095.799545779] [controller_manager]: Loading controller 'joint_state_broadcaster'
1678376095.8100915 [spawner-5] [INFO] [1678376095.809854318] [spawner_joint_impedance_controller]: [94mLoaded [1mjoint_impedance_controller[0m
1678376095.8257384 [spawner-6] [INFO] [1678376095.825405834] [spawner_joint_state_broadcaster]: [94mLoaded [1mjoint_state_broadcaster[0m
1678376095.8272803 [rox_control_node-1] [INFO] [1678376095.827105517] [controller_manager]: Configuring controller 'joint_state_broadcaster'
1678376095.8275547 [rox_control_node-1] [INFO] [1678376095.827242192] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
1678376095.8940248 [spawner-5] [INFO] [1678376095.893787919] [spawner_joint_impedance_controller]: [96mLoaded parameters file "/home/ros2/ws_spirit/install/kuka_rox_hw_interface/share/kuka_rox_hw_interface/config/joint_impedance_controller_config.yaml" for [1mjoint_impedance_controller[0m
1678376095.8943746 [spawner-5] [INFO] [1678376095.894141864] [spawner_joint_impedance_controller]: Loaded /home/ros2/ws_spirit/install/kuka_rox_hw_interface/share/kuka_rox_hw_interface/config/joint_impedance_controller_config.yaml into joint_impedance_controller
1678376095.8950522 [rox_control_node-1] [INFO] [1678376095.894917865] [controller_manager]: Configuring controller 'joint_impedance_controller'
1678376095.8974588 [spawner-5] Set parameter joints successful
1678376095.9738901 [INFO] [spawner-6]: process has finished cleanly [pid 20513]
1678376096.0183399 [rox_control_node-1] [INFO] [1678376096.018006449] [controller_manager]: Loading controller 'effort_controller'
1678376096.0247312 [spawner-7] [INFO] [1678376096.024540409] [spawner_effort_controller]: [94mLoaded [1meffort_controller[0m
1678376096.0320768 [spawner-7] [INFO] [1678376096.031883254] [spawner_effort_controller]: [96mLoaded parameters file "/home/ros2/ws_spirit/install/kuka_rox_hw_interface/share/kuka_rox_hw_interface/config/effort_controller_config.yaml" for [1meffort_controller[0m
1678376096.0324244 [spawner-7] [INFO] [1678376096.032213767] [spawner_effort_controller]: Loaded /home/ros2/ws_spirit/install/kuka_rox_hw_interface/share/kuka_rox_hw_interface/config/effort_controller_config.yaml into effort_controller
1678376096.0330479 [rox_control_node-1] [INFO] [1678376096.032885108] [controller_manager]: Configuring controller 'effort_controller'
1678376096.0335348 [rox_control_node-1] [INFO] [1678376096.033365241] [effort_controller]: configure successful
1678376096.0339060 [rox_control_node-1] got gains for joint_a1 as (5.000000, 0.000000, 0.100000)
1678376096.0340159 [rox_control_node-1] got gains for joint_a2 as (5.000000, 0.000000, 0.100000)
1678376096.0341234 [rox_control_node-1] got gains for joint_a3 as (5.000000, 0.000000, 0.100000)
1678376096.0342226 [rox_control_node-1] got gains for joint_a4 as (5.000000, 0.000000, 0.100000)
1678376096.0343087 [rox_control_node-1] got gains for joint_a5 as (5.000000, 0.000000, 0.100000)
1678376096.0343895 [rox_control_node-1] got gains for joint_a6 as (5.000000, 0.000000, 0.100000)
1678376096.0348134 [spawner-7] Set parameter joints successful
1678376096.0348995 [spawner-7] Set parameter gains.joint_a1.p successful
1678376096.0349646 [spawner-7] Set parameter gains.joint_a1.i successful
1678376096.0350225 [spawner-7] Set parameter gains.joint_a1.d successful
1678376096.0350804 [spawner-7] Set parameter gains.joint_a2.p successful
[... 24 lines cut ...]
1678376096.0662417 [spawner-4] Set parameter joints successful
1678376096.0663369 [spawner-4] Set parameter command_interfaces successful
1678376096.0663784 [spawner-4] Set parameter state_interfaces successful
1678376096.0664136 [spawner-4] Set parameter state_publish_rate successful
1678376096.0664463 [spawner-4] Set parameter action_monitor_rate successful
1678376096.1906300 [INFO] [spawner-7]: process has finished cleanly [pid 20515]
1678376096.2113073 [INFO] [spawner-4]: process has finished cleanly [pid 20509]
1678376106.6389141 [rox_control_node-1] [INFO] [1678376106.637892812] [resource_manager]: 'configure' hardware 'iisy_hardware' 
1678376106.6394484 [rox_control_node-1] [ERROR] [1678376106.638781104] [KukaRoXHardwareInterface]: SetQoSProfile failed
1678376106.6405642 [rox_control_node-1] [INFO] [1678376106.638840063] [resource_manager]: Failed to 'configure' hardware 'iisy_hardware'
1678376106.6409507 [robot_manager_node-2] [ERROR] [1678376106.639119470] [robot_manager]: Could not configure hardware interface
[... cut ...]
1678376114.7506933 [INFO] [robot_state_publisher-3]: process has finished cleanly [pid 20507]
1678376114.7508323 [INFO] [robot_manager_node-2]: process has finished cleanly [pid 20505]
1678376114.7509809 [INFO] [rox_control_node-1]: process has finished cleanly [pid 20503]
1678376114.7512724 [robot_state_publisher-3] [INFO] [1678376113.802612500] [rclcpp]: signal_handler(signum=2)
1678376114.7515290 [rox_control_node-1] [INFO] [1678376113.802612543] [rclcpp]: signal_handler(signum=2)
```

More information can be provided.

altex111, 2023-03-10:
Hi!

I tried to replicate the "SetQoSProfile failed error" message you encountered with.
For me it worked properly with the MOCK_KUKA_LIBS flag set to TRUE in the _kuka_rox_hardware_interface/CMakeLists.txt_. 

> No hardware is involved.

According to this you are not using any hardware so the flag should set to TRUE.

> SetQoSProfile failed

According the error message you sent the flag is set to FALSE.
Is my assumption correct?

Svastits, 2023-03-16:
Conclusion:

MOCK_KUKA_LIBS flag should be renamed to MOCK_HARDWARE (or something like that), and a comment should indicate in the cmakelists that additional packages are needed to build the non-mock version

closed 2023-05-19 by altex111