kroshu/kuka_drivers #94: Add joint impedance support for FRI

Svastits, 2023-06-30:
The hardware IF currently does not have stiffness and damping interfaces.
The manager node's control mode and command mode parameters could also be combined

alexswerner, 2023-08-18:
Hi, using the build in joint impedance controller would be great in my opinion. Even better would be to supply a example setup with a fully features Cartesian Impedance controller using the torque interface and disabling the internal joint impedance controller. Just joint impedance is IMHO not so useful. Maybe https://github.com/matthias-mayr/Cartesian-Impedance-Controller could be a start? I would be tentatively interesting in porting this to ROS2.

Svastits, 2023-08-21:
The description was indeed not really telling, but we plan to support both joint impedance and cartesian impedance control modes (with internal calculations). 
Joint impedance could be interesting for researchers, who would e.g want to try out cartesian impedance mode with their own IK, which might be a valid scenario with the 7-DOF robots IMHO.

So according to our plans, the following modes will be supported by our driver:
- joint position
- joint impedance
- joint torque (superimposed on internally calculated model torques)
- wrench
- cartesian pose
- cartesian impedance

(With FRI it is also possible to command axis positions with cartesian impendance attributes, or cartesian poses with joint impedance attributes, but I think we will not support these.)

So in the end, if one would not want to use the internal impedance calculations, he/she could use the linked controller + torque mode.

Do you think these modes cover all use cases of the driver?

alexswerner, 2023-08-21:
Certainly that sounds like a good approach. If the driver does all this, one would not need to set up something custom. From the Cartesian Impedance setup i would hope to get:
* Commandable from a JointTrajectoryController Interface or a Desired Frame Interface. 
* If using JointTrajectoryController interface, the Nullspace controller should track the desired joint configuration
* Overlay wrenches for applying a temporary force when running the cartesian impedance controller
* Smooth stiffness ramping (avoid jumps in command)
* Support both, asynchonous workflow (JointTrajectoryController) but also streaming/playback of desired frame/force from a topic. Ideally with sanitizing filters (position/velocity/acceleration)
* Tool frame offset setup: ros parameters - able to change this online
* Pickup Object weight/inertia for gravity compensation. Smooth transition when changing this.
* Stiffness configuration in tool frame, but ideally also world frame aligned stiffness (rotated frame). Damping as damping coefficients, not natural units (N/(m/s)).
* Saturated stiffness mode: Maximum force/torque
* Cartesian Impedance controller with Integral term to be able to run really low stiffness with acceptable static error.

Look like a wishlist. Please take is as one. Let me know if you have any questions. I'm happy to help.

And when you say FRI, would this be limited to Sunrise. Now we have an iiwa but will get an iisy soon. The same things would be nice over there as well.

Svastits, 2023-08-22:
Thanks for this list, it really helps us to know what the users would want :)
FRI currently supports cartesian impedance mode, whereas ECI (the equivalent for iisy) does not yet. In the future, ECI will also be extended with cartesian impedance mode, but that is a little more long term I would say