kroshu/kuka_drivers #98: Add support for cartesian motions

Svastits, 2023-07-31:
Cartesian position and wrench control are supported by FRI, but not by our driver, these functionalities should be added -> also to the application running on the controller

gavanderhoorn, 2023-07-31:
:+1: 

Would you already have an idea about the ROS API you'd want to use for this?

Ideally this would reuse an existing message set, to avoid further fragmentation and increase reuse.

I have a candidate set in mind, but I didn't want to immediately post it before seeing what you might have in mind.

Svastits, 2023-08-01:
No, we have not thought about the ROS API, at first we will focus on the robot controller side to enable to the cartesian types.
We would be glad if you could share your idea :)