kroshu/kuka_drivers #100: Cannot start controller for iiwa7

alexswerner, 2023-08-09:
Hello,

I have a problem starting the ros2 driver when working with an iiwa robot. Is this a way to get some support on this? If not, just let me know.

Setup:
ros2 humble (current)
ros2_kuka_drivers v0.0.2-402-gf93987c
FRI 1.15 on iiwa 14 R820

I'm able to run FRI Demo applications, e.g. "JointSineOverlay" . 
To reproduce the bug, first I start the different control nodes like this:
```
> ros2 launch kuka_sunrise lbr_iiwa7_control.launch.py
```
Trigger configuration
```
> ros2 lifecycle set /robot_manager configure
Transitioning failed
```
Output of `ros2 launch`
```
[control_node-1] [INFO] [1691617776.610808798] [resource_manager]: 'configure' hardware 'iiwa_hardware'
[control_node-1] [INFO] [1691617776.610896167] [resource_manager]: Successful 'configure' of hardware 'iiwa_hardware'
[robot_manager_node-2] [INFO] [1691617776.616033648] [robot_manager]: Successfully set 'controller_ip' parameter
[robot_manager_node-2] Request timed out
[robot_manager_node-2] [ERROR] [1691617779.616487786] [robot_manager]: Could not start controllers
[robot_manager_node-2] [INFO] [1691617779.618349611] [robot_manager]: Successfully connected to FRI
[robot_manager_node-2] Request timed out
[robot_manager_node-2] [ERROR] [1691617781.619562099] [robot_manager]: Could not stop controllers
```

On the smartPAD the output is:
```
Waiting for connection...
Connection established.
New data received: 01, [B@83969e 
Command received: CONNECT
Command executed.
New data received: 02, [B@1159092
Command receivda?: DISCONNECT
Command executed.
Error: connection lost. FRI ended.
Waiting for connection...
```

I've investigate in the source why this happens. Right now it is not clear to me why the controller sends DISCONNECT.

Svastits, 2023-08-10:
Hi, 
We could reproduce this bug on a VM, so we will try to take a look into this, but our capacity regarding the sunrise driver is a little limited at the moment. 
With a bare, RT-patched linux machine, the driver worked for as it should, we will try to grasp the differences between the two setups.
What client machine were you using when this bug occured? We could also not decode the driver version you used (ros2_kuka_drivers v0.0.2-402-gf93987c), is it the release tagged with v0.0.2?

alexswerner, 2023-08-10:
Hi,
thanks for the response. I'm running this on a Ubuntu 22.04 machine with ROS2 humble - using fastrtps as ROS2 middleware. System is a Intel(R) Core(TM) i7-8700T CPU, Hyperthreading is on, not limited to CPU cstate=1. Right now the system is also running a kernel without prempt-rt, but I can switch to one in case that matters at this step. RT-prios are available nonetheless.
The git version is from `git describe`. Unclear why is states v0.0.2 there. But the commit is correct f93987c so current master or one commit after the v0.0.7 tag.
I'm working on this right now and will provide another update in a couple of minutes.

alexswerner, 2023-08-10:
Now I assertained that `Request timed out` is actually the service call to `/controller_manager/switch_controllers` which does not return. And to me this is because there is a deadlock on the code.

 * In [control_node.cpp#59](https://github.com/kroshu/kroshu_ros2_core/blob/437f1c453bbbe79e6a7ac6a27635e5d6e9964560/src/control_node.cpp#L61) `update` is not called when `is_configure` is not set. 
 * Calling `update` however is necessary to perform the controller switch [controller_manager.cpp#L1930](https://github.com/ros-controls/ros2_control/blob/036640fe100fc9b5bb112a222618425db1cfd933/controller_manager/src/controller_manager.cpp#L1930). 
 * But again is_configure is set after switching the controllers in [robot_manager_nodes.cpp#127](https://github.com/kroshu/ros2_kuka_drivers/blob/f93987cda2ece0faf780390d17a0baf05951287c/kuka_sunrise_driver/src/robot_manager_node.cpp#L127). The call to `switch_controllers` is issued in the same file, but above at line 96. 

Maybe that helps in tracking it down.

Svastits, 2023-08-10:
Thanks for pointing this out, we indeed fixed this issue once in the commit you linked, but missed the change when combining the control nodes...
We will fix this bug tomorrow :)

Svastits, 2023-08-11:
We modified the controller node in kroshu_ros2_core to call the controller_manager's update even in unconfigured state.(https://github.com/kroshu/kroshu_ros2_core/pull/19) This indeed fixed the bug, I have tested it with all latest versions (now with a cleaned up workspace)
Please close the issue, if you can start the driver without problems :)
I just wanted to add a side note, that the urdf files for the iiwa were not yet verified, we will try to verify it and also add the urdf for the iiwa14
Thanks again for the issue and the proposed solution :)

alexswerner, 2023-08-18:
I can confirm this is fixed. Thanks!

closed 2023-08-18 by alexswerner