kroshu/kuka_drivers #101: ECI Mock driver not starting in Docker Container

morhof123, 2023-09-18:
Hi there,

I am currently trying to get the ECI driver running in a Docker Container.
I am following [the tutorial from the docs](https://github.com/kroshu/ros2_kuka_drivers/wiki#iiqka-driver-eci) to start the ECI driver in a mocked test environment to control the simulated robot with ROS.

## TL:DR
When trying to run `ros2 launch kuka_rox_hw_interface startup.launch.py`:
```text
...
[control_node-1] [INFO] [1695046574.399726173] [resource_manager]: Loading hardware 'LBRiisy3R760' 
[control_node-1] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[control_node-1]   what():  Failed to load library /custom_ws/install/kuka_rox_hw_interface/lib/kuka_rox_hw_interface/libkuka_rox_hw_interface.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libos-core-udp-communication.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
...
```

## Setup
- Ubuntu 20.04
- Docker version 24.0.6
- KUKA libs are mocked (`set(MOCK_KUKA_LIBS TRUE)`)

## Steps to reproduce
- Copy Dockerfile below
- Run `docker build -t eci_issue .`
- Enter Docker container via `docker run -it eci_issue /bin/bash` (if you want a display passthrough add `-v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY ` and on the host execute `xhost +`)
- Execute the command above `ros2 launch kuka_rox_hw_interface startup.launch.py`

## Files and Outputs
This is my Dockerfile (currently using the default ros:humble entrypoint)
```Dockerfile
FROM ros:humble

RUN apt-get update &&  apt-get install -y \
    libnanopb-dev \
    ros-humble-controller-manager \
    ros-humble-moveit \
    ros-humble-moveit-visual-tools \
    ros-humble-joint-trajectory-controller \
    ros-humble-joint-state-broadcaster \
    ros-humble-joint-state-publisher \
    ros-humble-ros2-controllers \
    libprotobuf-dev \
    protobuf-compiler \
    libprotoc-dev \
    libgrpc++-dev \
    byobu \
 && rm -rf /var/lib/apt/lists/* # byobu is like tmux only for testing

RUN mkdir -p /custom_ws/src

WORKDIR /custom_ws/src

RUN git clone https://github.com/kroshu/kroshu_ros2_core.git
RUN git clone https://github.com/kroshu/ros2_kuka_drivers.git
RUN git clone https://github.com/kroshu/kuka_simulators.git

WORKDIR /custom_ws

RUN /bin/bash -c "source /opt/ros/humble/setup.bash && colcon build"
RUN /bin/bash -c "echo 'source /custom_ws/install/setup.bash' >> ~/.bashrc"
```

This should install all dependencies and successfully builds.
It is possible to start the rsi simulation, but when I want to start the eci startup.launch.py file it appears not to be working:

Console output when running `ros2 launch kuka_rox_hw_interface startup.launch.py`:
```text
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-09-18-14-16-13-840313-5c6f4854484c-53611
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [control_node-1]: process started with pid [53642]
[INFO] [robot_manager_node-2]: process started with pid [53644]
[... 51 lines cut ...]
[spawner-8] [ERROR] [1695046585.272609936] [spawner_control_mode_handler]: Controller manager not available
[spawner-6] [ERROR] [1695046585.273111428] [spawner_joint_impedance_controller]: Controller manager not available
[ERROR] [spawner-8]: process has died [pid 53656, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner control_mode_handler -c /controller_manager -p --inactive --ros-args'].
[ERROR] [spawner-6]: process has died [pid 53652, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_impedance_controller -c /controller_manager -p /custom_ws/install/kuka_rox_hw_interface/share/kuka_rox_hw_interface/config/joint_impedance_controller_config.yaml --inactive --ros-args'].
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_manager_node-2] [INFO] [1695046604.619778206] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 53646]
[INFO] [robot_manager_node-2]: process has finished cleanly [pid 53644]
[robot_state_publisher-3] [INFO] [1695046604.619803602] [rclcpp]: signal_handler(signum=2)
```

Output, when trying to run `ros2 lifecycle set robot_manager configure`:
```text
root@d354a4fe0b00:/custom_ws# ros2 lifecycle set robot_manager configure
Transitioning failed
[robot_manager_node-2][ERROR] [robot_manager]: Could not configure hardware interface
```

As you can see, the control node throws an exception when Loading the Mock library for the udp communication, which probably is the point of failure here. But when we look into the CMakeLists [the library should be added.](https://github.com/kroshu/ros2_kuka_drivers/blob/e1ed35ce9bf48b89100d0dd22530df68da553952/kuka_rox_hardware_interface/CMakeLists.txt#L49)

Do you also encounter this error or have an idea how to fix it?

kind regards

Moritz

Svastits, 2023-09-18:
Hi,

What ROS version are you using? We have tested this on ROS2 humble (Ubuntu 22.04), where no such issues occured. My first guess is that Ubuntu 20.04 has some incompabilities and that is the cause of this issue

morhof123, 2023-09-19:
I will try to spin up a VM and try it out in Ubuntu 22.04, but IMO it shouldn't matter as everything is being built inside the official ros:humble Docker container.
I will post an update if it worked :+1:

morhof123, 2023-09-19:
I found a possible solution:
When running `echo $LD_LIBRARY_PATH ` we can see, that the kuka_rox_hw_interface is not in our library path:
```
/custom_ws/install/kuka_rsi_hw_interface/lib:/custom_ws/install/kuka_driver_interfaces/lib:/custom_ws/install/kroshu_ros2_core/lib:/custom_ws/install/fri/lib:/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
```

I then manually installed the `os-core-udp-communication` by adding the following config to [`/custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists.txt` after line 88:](https://github.com/kroshu/ros2_kuka_drivers/blob/e1ed35ce9bf48b89100d0dd22530df68da553952/kuka_rox_hardware_interface/CMakeLists.txt#L88)
```CMake
...
install(TARGETS os-core-udp-communication
LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR}
PUBLIC_HEADER DESTINATION ${CMAKE_INSTALL_INCLUDEDIR})
...
```

Now after resourcing the setup.bash file the error disappears and the path is added to the LD_LIBRARY_PATH:
```text
root@90b7c46baa71:/custom_ws# source install/setup.bash 
root@90b7c46baa71:/custom_ws# echo $LD_LIBRARY_PATH 
/custom_ws/install/kuka_rox_hw_interface/lib:/custom_ws/install/kuka_rsi_hw_interface/lib:/custom_ws/install/kuka_driver_interfaces/lib:/custom_ws/install/kroshu_ros2_core/lib:/custom_ws/install/fri/lib:/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
```

morhof123, 2023-09-19:
For anyone wondering how the now working Dockerfile looks like, here it is:

```Dockerfile
FROM ros:humble

RUN apt-get update &&  apt-get install -y \
    libnanopb-dev \
    ros-humble-controller-manager \
    ros-humble-moveit \
    ros-humble-moveit-visual-tools \
    ros-humble-joint-trajectory-controller \
    ros-humble-joint-state-broadcaster \
    ros-humble-joint-state-publisher \
    ros-humble-ros2-controllers \
    libprotobuf-dev \
    protobuf-compiler \
    libprotoc-dev \
    libgrpc++-dev \
    byobu \
 && rm -rf /var/lib/apt/lists/* # byobu is like tmux only for testing

RUN mkdir -p /custom_ws/src

WORKDIR /custom_ws/src

RUN git clone https://github.com/kroshu/kroshu_ros2_core.git
RUN git clone https://github.com/kroshu/ros2_kuka_drivers.git
RUN git clone https://github.com/kroshu/kuka_simulators.git
RUN git clone https://github.com/kroshu/ros2_kuka_controllers.git

WORKDIR /custom_ws

# Custom fix for now
RUN cp /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists.txt /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists_old.txt
RUN awk 'NR==88{print "install(TARGETS os-core-udp-communication LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR} PUBLIC_HEADER DESTINATION ${CMAKE_INSTALL_INCLUDEDIR})"}1' /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists_old.txt > /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists.txt

RUN /bin/bash -c "source /opt/ros/humble/setup.bash && colcon build"
RUN /bin/bash -c "echo 'source /custom_ws/install/setup.bash' >> ~/.bashrc"
...
```
If this issues is fixed in the future, it shouldn't be necessarry to need the two lines below `# Custom fix for now`

Svastits, 2023-09-19:
Thanks for investigating this issue :)
We still could not reproduce it on VM-s or native Linux computers, but indeed the kuka_rox_hw_interface is missing from the $LD_LIBRARY_PATH. I fixed this in this branch by exporting the library in cmake: 
https://github.com/kroshu/ros2_kuka_drivers/tree/fix/ld_library_path

Could you check if this solved your issue too?

morhof123, 2023-09-20:
You are right, the kuka_rox_hw_interface is now in $LD_LIBRARY_PATH, but the same error persists, as the library os-core-udp-communication is not in the kuka_rox_hw_interface folder:

```text
root@ros_test:/custom_ws# echo $LD_LIBRARY_PATH 
/custom_ws/install/kuka_rox_hw_interface/lib:/custom_ws/install/kuka_rsi_hw_interface/lib:/custom_ws/install/kuka_controllers/lib:/custom_ws/install/kuka_driver_interfaces/lib:/custom_ws/install/kroshu_ros2_core/lib:/custom_ws/install/fri/lib:/custom_ws/install/effort_controllers/lib:/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib

root@ros_test:/custom_ws# ls /custom_ws/install/kuka_rox_hw_interface/lib/kuka_rox_hw_interface/
libkuka_rox_hw_interface.so  robot_manager_node

root@ros_test:/custom_ws# ls /custom_ws/install/kuka_rox_hw_interface/lib/
kuka_rox_hw_interface
```

Maybe you can't reproduce the error, because the os-core-udp-communication library is already in your LD_LIBRARY_PATH somewhere else?
It only works on my side if I add an install of the os-core-udp-communictaion like before.
Have you tried using the Dockerfile to recreate the issue?

Here is a Dockerfile using the fix/ld_library_path branch, just copy the contents inside a file named `Dockerfile` and in the same directory run `docker build -t eci_issue .` and to start it `docker run eci_issue`:
```Dockerfile
FROM ros:humble

RUN apt-get update &&  apt-get install -y \
    libnanopb-dev \
    ros-humble-controller-manager \
    ros-humble-moveit \
    ros-humble-moveit-visual-tools \
    ros-humble-joint-trajectory-controller \
    ros-humble-joint-state-broadcaster \
    ros-humble-joint-state-publisher \
    ros-humble-ros2-controllers \
    libprotobuf-dev \
    protobuf-compiler \
    libprotoc-dev \
    libgrpc++-dev \
 && rm -rf /var/lib/apt/lists/* 

RUN mkdir -p /custom_ws/src

WORKDIR /custom_ws/src

RUN git clone https://github.com/kroshu/kroshu_ros2_core.git
RUN git clone https://github.com/kroshu/ros2_kuka_drivers.git -b fix/ld_library_path
RUN git clone https://github.com/kroshu/kuka_simulators.git
RUN git clone https://github.com/kroshu/ros2_kuka_controllers.git

WORKDIR /custom_ws

# Custom fix for now
#RUN cp /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists.txt /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists_old.txt
#RUN awk 'NR==88{print "install(TARGETS os-core-udp-communication LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR} PUBLIC_HEADER DESTINATION ${CMAKE_INSTALL_INCLUDEDIR})"}1' /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists_old.txt > /custom_ws/src/ros2_kuka_drivers/kuka_rox_hardware_interface/CMakeLists.txt

RUN /bin/bash -c "source /opt/ros/humble/setup.bash && colcon build"
RUN /bin/bash -c "echo 'source /custom_ws/install/setup.bash' >> ~/.bashrc"

RUN /bin/bash -c "echo 'source /opt/ros/humble/setup.bash && source  /custom_ws/install/setup.bash && ros2 launch kuka_rox_hw_interface startup.launch.py' > /custom_ws/startup.bash"
RUN chmod +x /custom_ws/startup.bash

ENTRYPOINT ["/bin/bash", "/custom_ws/startup.bash"]
```

Svastits, 2023-09-20:
Yesterday I used your Dockerfile and the fix seemed to have resolved the issue, but I will look at it again then :)

I think I have the original (non-mock) version of the udp lib installed on my PC, that could indeed explain why the issue was not there for me

Svastits, 2023-09-21:
I think I found the reason for not being able to reproduce this bug previously: I always use the --symlink-install flag for colcon build, which installs a symbolic link of the .so file located in the build folder. This .so has the udp lib right next to it so the library loader can find it.

So the quickest fix would be to modify your docker file to use --symlink-install, but I will also modify cmake soon to fix the "normal" build

morhof123, 2023-09-21:
> So the quickest fix would be to modify your docker file to use --symlink-install, but I will also modify cmake soon to fix the "normal" build

This seams to fix it, thank you for investigating :smile: 
I adjusted the Dockerfile (see below if anyone runs into the same problem).
<details>

```Dockerfile
FROM ros:humble

RUN apt-get update &&  apt-get install -y \
    libnanopb-dev \
    ros-humble-controller-manager \
    ros-humble-moveit \
    ros-humble-moveit-visual-tools \
    ros-humble-joint-trajectory-controller \
    ros-humble-joint-state-broadcaster \
    ros-humble-joint-state-publisher \
    ros-humble-ros2-controllers \
    libprotobuf-dev \
    protobuf-compiler \
    libprotoc-dev \
    libgrpc++-dev \
 && rm -rf /var/lib/apt/lists/* 

RUN mkdir -p /custom_ws/src

WORKDIR /custom_ws/src

RUN git clone https://github.com/kroshu/kroshu_ros2_core.git
RUN git clone https://github.com/kroshu/ros2_kuka_drivers.git -b fix/ld_library_path
RUN git clone https://github.com/kroshu/kuka_simulators.git
RUN git clone https://github.com/kroshu/ros2_kuka_controllers.git

WORKDIR /custom_ws

RUN /bin/bash -c "source /opt/ros/humble/setup.bash && colcon build --symlink-install"
RUN /bin/bash -c "echo 'source /custom_ws/install/setup.bash' >> ~/.bashrc"

RUN /bin/bash -c "echo 'source /opt/ros/humble/setup.bash && source  /custom_ws/install/setup.bash && ros2 launch kuka_rox_hw_interface startup.launch.py' > /custom_ws/startup.bash"
RUN chmod +x /custom_ws/startup.bash

ENTRYPOINT ["/bin/bash", "/custom_ws/startup.bash"]
```

Quick note, if fix/ld_library_path is merged into the main branch in the future, just remove the `-b fix/ld_library_path` flag on git clone

</details>

Svastits, 2023-09-25:
I modified the Cmakelists to install the udp library in mock mode.
The PR closed the issue, I will reopen it, please close it if it is fixed on your side too

morhof123, 2023-09-25:
Looks good to me, thank you so much for the support :smile:

closed 2023-09-25 by morhof123