kroshu/kuka_drivers #102: RSI Moveit Lauch

ClaBirk, 2023-09-24:
I want to integrate pilz planner into the launch file for the RSI HW Interface.
How can this be added ?

Svastits, 2023-09-25:
The launch file only starts the server side of moveit.
To use a specific planner, I think the easiest way is to specify the planner id in the motion planning request (on the client side, so the launch file does not need to be modified)
Example for this can be found here: https://github.com/kroshu/ros2_kuka_drivers/blob/e1ed35ce9bf48b89100d0dd22530df68da553952/kuka_driver_examples/eci_demo/src/MoveitBasicPlan.cpp#L66C1-L67C1

ClaBirk, 2023-09-25:
okay i got this launch file to launch my Robot in Moveit:
[Moveit_launch.txt](https://github.com/kroshu/ros2_kuka_drivers/files/12712362/Moveit_launch.txt)
Here i load the moveitconfig in order to be able to plan movements.

And this is an Cpp file wich i wrote to control the robot
[Control_with_moveit_cpp.txt](https://github.com/kroshu/ros2_kuka_drivers/files/12712394/Control_with_moveit_cpp.txt)

Here is my package for reference with the moveit_config included
[my_Package.zip](https://github.com/kroshu/ros2_kuka_drivers/files/12712406/my_Package.zip)

i already got this as a simulation for what i want to do with the robot now i want to transfer it to real hardware with your rsi hw_interface

Would be awesome if you could help me out

closed 2023-09-27 by Svastits

viraj-awate-work, 2026-09-03:
Is RSI module required for Moveit Control for KR16-1610 2 C4 with KSS 8.6.8?

Svastits, 2026-09-03:
yes, if you would like to stream positions to the robot, RSI is always required for KR robots

viraj-awate-work, 2026-09-03:
Thank you for replying. What other way is there for communicating Moveit to Kuka 16 - 1610 - 2 C4 with KSS 8.6.8 because RSI module is expensive and I want to do the control in a budget friendly way

Svastits, 2026-09-03:
technically you can connect moveit to EKI as well, but that is not yet implemented, so the only supported way is rsi

viraj-awate-work, 2026-09-03:
How different is that from current RSI because I am interested in writing it because of my work needs

Svastits, 2026-09-03:
RSI streams joint positions at 250Hz, which is connected with moveit&ros2_control by our driver.
for EKI, you have to write the KRL program, and technically you could stream trajectory segments at about 50Hz, that your KRL code could process, but the effort required for this is not negligable