kroshu/kuka_drivers #105: Interface starts but stops immediatly after that

ClaBirk, 2023-09-26:
[control_node-1] [INFO] [1695728579.810712266] [UDPServer]: 192.168.1.140: 49152
[control_node-1] [INFO] [1695728579.810831246] [KukaRSIHardwareInterface]: Connecting to robot . . .
[control_node-1] [INFO] [1695728585.019093222] [KukaRSIHardwareInterface]: Got data from robot
[control_node-1] [INFO] [1695728585.019692347] [KukaRSIHardwareInterface]: System Successfully started!
[control_node-1] [INFO] [1695728585.019737103] [resource_manager]: Successful 'activate' of hardware 'kr6_r700_sixx'
[control_node-1] [ERROR] [1695728586.022863020] [KukaRSIHardwareInterface]: No data received from robot

Svastits, 2023-09-27:
At first glance it seems to me that there is an internal RSI error.
The first state is published from RSI to you client, it sends a response, but than some error occurs.

The error code visible on the TP (KSS29002) is indeed an internal RSI error and can have two causes:
- RSI object configured incorrectly
- Software bug in RobotSensorInterface V3.x: The bug occurs if the object “POSCORRMON” is pasted in RSIVisual and RSI is started with “RSI_ON” in “#IPO” mode.

The 2. option should not occur, as you have no POSCORRMON object, so I assume the first is the case.
I think the issue is, that RSI 3.2 might not support the same rsix format as later versions, previously I think there were 2 file instead of the rsix: one .rsi.diagram and one .rsi.xml, and the system generated the rsix based on these.

Can you load the RSI context in WorkVisual? I am not sure how to generate a correct context file for your version, maybe you could try to create yor own context in your own environment.

It should look like this:
[image]

ClaBirk, 2023-09-27:
Thanks for your fast reply.
How to open the RSI context in WorkVisual? 
I prefer updating the RSI if possible - i contacted KUKA in this reguard and will keep you up to date weather this is an option or not.

Svastits, 2023-09-27:
The RSI context (.rsix file) should be visible in the files menu of your project (Config:\User\Common\SensorInterface\rox_rsi.rsix)
You will need an RSIVisual tool/option package to be able to see it visually

ClaBirk, 2023-09-28:
I had to install RSI Visual Shell wich was located at D/ opt 
Is a Robot from 2011

On Github i found these 2 Configs:

https://github.com/ros-industrial/kuka_experimental
[image]

https://github.com/cjlh/kuka-rsi-ros-interface
[image]

The ROS Industrial one looks verry similar but has no Stop

Updating to RSI 4.0 will require hardware updates as well. 

You have an idea what could go wrong with the RSI Config or a idea how to debug it?

Svastits, 2023-09-28:
We have ported our driver from the ros_industrial one and added the stop flag to be able to stop control without having to wait for the timeouts, so it is indeed similiar :)
I think this is a version incompatibility issue... can you try to add the stop block to your diagram and use that context instead of the .rsix one?

ClaBirk, 2023-09-28:
okay i will examin the structure and try to add it.
Does the missing stop in the rsi explain the error of the interface immediatly after startup?

ClaBirk, 2023-10-02:
[krc4_files.zip](https://github.com/kroshu/ros2_kuka_drivers/files/12780451/krc4_files.zip)

I found a working config for my RSI version. I changed the .rsi with rsi visual accordingly.
You can use these files for your repository if you like.

from here on i want to integrate my robot kr10_r1100_sixx with moveit to this package.

I think there is tow main steps to do:

1. integrate the kr10 r1100 in the package
  What do i need to change or add in order to do that? 
  I saw there are several nodes wich need robot description passed along so it doesnt seem to be plug and play
  
2. I got a working moveit config wich a launchfile - move group node must be running in order to plan trajectories. How to integrate that?

Svastits, 2023-10-02:
Thanks for the config files, I will try to place them somewhere to also have support for older RSI verisons :)

We would like a plug&play solution for moveit too, it is just not implemented yet for KSS robots, only for LBR iisy-s...
The other issue is, that the driver does not yet support the kr10r1000_sixx, so you will have to also edit the [launch file](https://github.com/kroshu/ros2_kuka_drivers/blob/master/kuka_rsi_hw_interface/launch/startup.launch.py) too...

[This](https://github.com/kroshu/ros2_kuka_drivers/blob/master/kuka_driver_examples/eci_demo/launch/moveit_planning_example.launch.py) launch file starts the driver with a robot model argument and the movegroup node, you could just modify this to follow your package names+ urdf/srdf descriptions

Starting the rsi hardware interface (ros2 launch kuka_rsi_hw_interface startup.launch.py + configuring and activating the lifecycle nodes) opens the interfaces needed for the rqt joint trajectory controller, so if you have added the kr10 to the supported robots in the launch file, that interface should be available to you.

ClaBirk, 2023-10-13:
Does the RSI Interface work with T2 mode as well ? so far i got it running with T1 mode?

closed 2023-10-13 by ClaBirk