kroshu/kuka_drivers #117: IO comfig handling from harware interface

altex111, 2023-11-17:


Svastits, 2024-01-08:
@altex111, as iiQKA will not support RT I/O-s, I will close this issue so as not to confuse users...

closed 2024-01-08 by Svastits

prasuchit, 2024-01-18:
@Svastits @altex111 We just purchased a KUKA iisy 3 R760 which (if I understand correctly) comes with the iiQKA OS and we're looking to purchase a thirdparty soft gripper. However, when I got in touch with KUKA customer service, they informed me that the ROS2 drivers don't support sending I/O signals to the end-effector link through the X96 and X94 connectors.

We would really appreciate it if you could add this feature to the KUKA iiQKA EAC driver package. If this is not planned anytime soon, kindly provide guidance on how I can send signals to the X96 and X94 connectors through either ROS or Python code.

Hoping to hear from you soon. Thanks.

@gavanderhoorn @Groebehavn @oprezz

Svastits, 2024-01-19:
Unfortunately the iiQKA ExternalAPI.Control option package does not support I/O control, therefore it is not possible to extend the driver...
The only solution I think is to connect the necessary signal bus(es) directly to the client computer and control the I/Os from there.

prasuchit, 2024-01-19:
Hi @Svastits, thank you for the prompt response. Could you please elaborate on the part "connect the necessary signal bus(es) directly to the client computer and control the I/Os from there"? Do you mean through an external ethercat interface? Do you happen to have any guide on how to do that?

Svastits, 2024-01-23:
Unfortunetaly I do not have much experience connecting I/O-s, and besides this should be quite hardware-specific... 
I assume the gripper should have some kind of manual describing how to connect to systems, you should follow that (the external ethercat interface is quite possible)
If you need help in connecting to the `ros2_control` framework, I can provide more help related to this (in short, you have to create a new hardware interface that can read and write data on the bus and load this to the control loop)