kroshu/kuka_drivers #119: possible typo in `kuka_drivers/examples/iiqka_moveit_example/include/iiqka_moveit_example/moveit_example.hpp`

uriel-luviano, 2023-11-28:
Hi! I am trying to build the project in order to control a KUKA KR4 robotic arm. Following the instructions in the readme the `colcon build` command gave me the following error:
```bash
In file included from /home/uriel/ws_moveit/src/kuka_drivers/examples/iiqka_moveit_example/src/MoveitCollisionAvoidanceExample.cpp:18:
/home/uriel/ws_moveit/src/kuka_drivers/examples/iiqka_moveit_example/include/iiqka_moveit_example/moveit_example.hpp: In member function ‘moveit_msgs::msg::RobotTrajectory_<std::allocator<void> >::SharedPtr MoveitExample::planToPoint(const Isometry3d&, const string&, const string&)’:
/home/uriel/ws_moveit/src/kuka_drivers/examples/iiqka_moveit_example/include/iiqka_moveit_example/moveit_example.hpp:109:71: error: ‘struct moveit::planning_interface::MoveGroupInterface::Plan’ has no member named ‘trajectory_’; did you mean ‘trajectory’?
  109 |       return std::make_shared<moveit_msgs::msg::RobotTrajectory>(plan.trajectory_);
      |                                                                       ^~~~~~~~~~~
      |                                                                       trajectory
```
Changing `trajectory_` to `trajectory` in lines 109, 125 and 150 seems to fix the issue and the build finishes successfully. I'm not a very experienced ROS user and I don't know if this is an actual fix or if the error was somewhere else.

I'm writing this issue to see if you can help me find out which is the case. Thanks in advance and apologies in case this is not the way to proceed.

Svastits, 2023-11-28:
What version of ROS are you using? The driver is for humble, but for iron the moveit API was changed which broke the build...

altex111, 2023-11-28:
I believe When you downloaded the moveit, you downloaded the iron versions, with the new API changes, as mentioned above.

I suggest you install moveit with apt install, follow the instructions here: https://moveit.ros.org/install-moveit2/binary/

make sure to delet every packege before installing the new ones, this command can help:
`sudo apt remove ros-$ROS_DISTRO-moveit*`

closed 2023-11-28 by uriel-luviano