kroshu/kuka_drivers #140: [help wanted] KUKA KMR iiwa

liyifei1997, 2024-01-16:
The autonomous KMR iiwa robot is HRC-capable and mobile. It combines the strengths of the sensitive LBR iiwa lightweight robot with those of a mobile, autonomous platform.

I'm wondering if it's possible to add support for the KMR iiwa omnimove LBR iiwa 14 R820.

Thanks so much.

Svastits, 2024-01-17:
Unfortunately we will not have the capacity to add support for KMR iiwas in the near future, but we will look into it when we have the resources.

gavanderhoorn, 2024-01-17:
We also have an IIWA on a KMR and would also be interested in proper ROS support for that combination.

(just to add my :+1: to this request)

mewescott, 2025-03-28:
Has the KMR been added yet.  I am working on ROS 2 humble implementation for a KMR Iowa right now.

liyifei1997, 2025-06-27:
> Sorry to hear that.  We have a real commercial use for it with ROS.  Would be nice if Kuka supported that.  I did get consistent jogging of the kmr to work yesterday.  I really need someone to help me figure out how to make the FDI connection give me Odom and laser data.
> […](#)

Hi, 
I'm curious to know which library you're using. I am currently using this one, [ROS KMR iiwa](https://github.com/stoic-roboticist/kmriiwa_ros_java) but it is not very user-friendly, and the arm and mobile base cannot be used together.

Botman-Mike, 2025-06-27:
> CAUTION: This email originated from outside of Spirit AeroSystems. Do not click links or open attachments unless you recognize the sender and know the content is safe.

That is a ROS 1 setup.  I started with the ROS2 one here:
 https://github.com/MortenMDahl/kmriiwa_ws_devel

It was originally developed by Nina Wahl and Charlotte Heggem as a master thesis project at NTNU.  This one was upgraded to Foxy and I upgraded to Humble along with several other improvements.

torydebra, 2026-10-14:
Hello,

Has the community found a solution for using the KMR base with ROS2 as today? 
I think that the known https://github.com/ninamwa/kmriiwa_ws and its various forks have been using an outdated version of the sunrise / navigation KUKA libraries