kroshu/kuka_drivers #150: [BUG] - iiQKA Communication with LBR iisy R1300

ekampourakis, 2024-03-13:
### Description

Cannot get communication between iiQKA and the robot driver possibly due to missing toolbox installation of ExternalAPI.Control
The documentation states as a first step to install the toolbox to the controller but the installation process is not documented and there are no results regarding the toolbox online.
Wireshark shows the traffic from the client to the controller after trying to configure the robot manager but gets no response.
Possibly the toolbox is not installed and the ports are closed.

### KUKA robot OS

iiQKA

### KUKA robot OS version

KUKA Linux OS 2.0.14

### KUKA external interface version

iiQKA OS 1.2.6

### Affected robot model(s)

LBR iisy R1300

### Version or commit hash of the driver

master

### Setup

```bash
The LBR iisy R1300 controller is freshly upgraded from the teach pendant.
Network is configured and accessible between the controller and client PC
Client is on 192.168.10.102 and controller on 192.168.10.100.
All 3 ROS2 packages are built and the external SDK is build prior to that.
The test was performed both in T1 and AUT and as User and Admin.
```

### Reproduction steps

```bash
colcon build
source install/setup.bash
ros2 launch kuka_iiqka_eac_driver startup.launch.py client_ip:=192.168.10.102 controller_ip:=192.168.10.100
Release SPOC on Teach Pendant
ros2 lifecycle set robot_manager configure
```

### Logs

```bash
[INFO] [launch]: All log files can be found below /home/ekampourakis/.ros/log/2024-03-13-11-24-41-600258-ekampourakis-pc-15372
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [control_node-1]: process started with pid [15385]
[INFO] [robot_manager_node-2]: process started with pid [15387]
[INFO] [robot_state_publisher-3]: process started with pid [15389]
[INFO] [spawner-4]: process started with pid [15391]
[INFO] [spawner-5]: process started with pid [15393]
[INFO] [spawner-6]: process started with pid [15395]
[INFO] [spawner-7]: process started with pid [15397]
[INFO] [spawner-8]: process started with pid [15399]
[INFO] [spawner-9]: process started with pid [15401]
[robot_manager_node-2] [INFO] [1710321881.814935552] [robot_manager]: Control mode change requested
[robot_manager_node-2] [INFO] [1710321881.814969437] [robot_manager]: Successfully changed control mode to 1
[robot_state_publisher-3] [INFO] [1710321881.817378138] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1710321881.817417704] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1710321881.817422060] [robot_state_publisher]: got segment flange
[robot_state_publisher-3] [INFO] [1710321881.817424690] [robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1710321881.817427028] [robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1710321881.817429247] [robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1710321881.817431342] [robot_state_publisher]: got segment link_4
[... 32 lines cut ...]
[spawner-4] [INFO] [1710321882.311023391] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[control_node-1] [INFO] [1710321882.314485137] [controller_manager]: Configuring controller 'joint_trajectory_controller'
[control_node-1] [INFO] [1710321882.314569534] [joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[control_node-1] [INFO] [1710321882.314579151] [joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position].
[control_node-1] [INFO] [1710321882.314584101] [joint_trajectory_controller]: Using 'splines' interpolation method.
[spawner-8] [INFO] [1710321882.314874339] [spawner_control_mode_handler]: Loaded control_mode_handler
[control_node-1] [INFO] [1710321882.314883357] [joint_trajectory_controller]: Controller state will be published at 50.00 Hz.
[control_node-1] [INFO] [1710321882.315850127] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[control_node-1] [INFO] [1710321882.318542760] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[control_node-1] [INFO] [1710321882.318569939] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[control_node-1] [INFO] [1710321882.322498885] [controller_manager]: Configuring controller 'control_mode_handler'
[control_node-1] [INFO] [1710321882.323566965] [control_mode_handler]: Control mode handler configured
[INFO] [spawner-7]: process has finished cleanly [pid 15397]
[INFO] [spawner-6]: process has finished cleanly [pid 15395]
[INFO] [spawner-4]: process has finished cleanly [pid 15391]
[INFO] [spawner-8]: process has finished cleanly [pid 15399]
[INFO] [spawner-5]: process has finished cleanly [pid 15393]
[INFO] [spawner-9]: process has finished cleanly [pid 15401]

[control_node-1] [INFO] [1710322015.382291688] [resource_manager]: 'configure' hardware 'lbr_iisy11_r1300' 
[control_node-1] [INFO] [1710322015.382369347] [control_mode_handler]: Control mode changed to 1
[robot_manager_node-2] [ERROR] [1710322017.382186975] [robot_manager]: Could not configure hardware interface
[control_node-1] [ERROR] [1710322035.383135890] [KukaEACHardwareInterface]: QoS configuration failed, error message: failed to connect to all addresses
[control_node-1] [INFO] [1710322035.383167251] [resource_manager]: Failed to 'configure' hardware 'lbr_iisy11_r1300'
```

Svastits, 2024-03-13:
The ExternalAPI.Control toolbox is not available by default, you have to request it from KUKA customer support. After installation all the necessary ports will be open and the driver should work.

I will extend the doc with this information

ekampourakis, 2024-03-13:
I contacted support and will report back once the issue gets resolved. Feel free to close the issue or leave it open for further validation

closed 2024-03-13 by Svastits