kroshu/kuka_drivers #166: [BUG] - "No kinematics plugins defined. Fill and load kinematics.yaml!" MoveIt planning keeps failing.

prasuchit, 2024-05-24:
### Description

Hi @Svastits,

I'm currently using kuka_drivers/iiqka_moveit_example/moveit_planning_example.launch.py with use_fake_hardware:=true.

I'm able to launch successfully, see the robot with the tf frames after I do configure and activate and plan and execute motions within moveit2-rviz.

The problem begins when I start trying to programmatically plan motions. Every time I run my node, I see this warning:

[WARN] [1716557929.990758924] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!

and eventually, it just times out and fails to plan:

[ERROR] [1716559618.572070969] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
[ERROR] [1716559618.572111417] [first_test]: Planning failed!

I have tried many different poses and tried giving it more time to plan, etc. This is what my code looks like:

    int main(int argc, char *argv[]) {
        // Initialize ROS and create the Node
        rclcpp::init(argc, argv);
        auto const node = std::make_shared<rclcpp::Node>(
            "first_test",
            rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(
                true));
        // We spin up a SingleThreadedExecutor for the current state monitor to get information
        // about the robot's state.
        rclcpp::executors::SingleThreadedExecutor executor;
        executor.add_node(node);
        std::thread([&executor]() { executor.spin(); }).detach();
    
        // Create a ROS logger
        auto const logger = rclcpp::get_logger("first_test");
    
        static const std::string PLANNING_GROUP = "manipulator";
    
        // Create the MoveIt MoveGroup Interface
        using moveit::planning_interface::MoveGroupInterface;
        auto move_group = MoveGroupInterface(node, PLANNING_GROUP);
     
        // Create planning request using RRTConnectkConfigDefault motion planner
        move_group.setPlanningPipelineId("ompl");
        move_group.setPlannerId("RRTConnectkConfigDefault");
        std::copy(move_group.getJointModelGroupNames().begin(), move_group.getJointModelGroupNames().end(),
                std::ostream_iterator<std::string>(std::cout, ", "));
        moveit::core::RobotState start_state(*move_group.getCurrentState());
        // Set a target Pose
        // 0.387434 -0.022426 0.320466 0.038751 0.996789 -0.019382 0.067334
        auto const target_pose = [] {
            geometry_msgs::msg::Pose msg;
            msg.orientation.x = 0.38751;
            msg.orientation.y = 0.996789;
            msg.orientation.z = -0.019382;
            msg.orientation.w = 0.067334;
            msg.position.x = 0.3874;
            msg.position.y = -0.022426;
            msg.position.z = 0.3205;
            return msg;
        }();

        move_group.setStartState(start_state);
        move_group.setPoseTarget(target_pose);
        move_group.setPlanningTime(20.0);
        // Create a plan to that target pose
        auto const [success, plan] = [&move_group] {
            moveit::planning_interface::MoveGroupInterface::Plan msg;
            auto const ok = static_cast<bool>(move_group.plan(msg));
            return std::make_pair(ok, msg);
        }();
    
        // Execute the plan
        if (success) {
            move_group.execute(plan);
        } else {
            RCLCPP_ERROR(logger, "Planning failed!");
        }
    
        // print current pose
        geometry_msgs::msg::Pose current_pose = move_group.getCurrentPose().pose;
    
        // Print the current pose of the end effector
        RCLCPP_INFO(node->get_logger(), "Current pose: %f %f %f %f %f %f %f",
            current_pose.position.x,
            current_pose.position.y,
            current_pose.position.z,
            current_pose.orientation.x,
            current_pose.orientation.y,
            current_pose.orientation.z,
            current_pose.orientation.w);
    
        // Shutdown ROS
        rclcpp::shutdown();
        return 0;
    }
    
    
As you can see, it is quite simple and straightforward. I see that the kinematics.yaml is present [here](https://github.com/kroshu/kuka_robot_descriptions/blob/master/kuka_lbr_iisy_moveit_config/config/kinematics.yaml) and being loaded into rosparam [here](https://github.com/kroshu/kuka_drivers/blob/e024a00b97783126d0cd61e1a98b7db343d4fa30/examples/iiqka_moveit_example/launch/moveit_planning_example.launch.py#L61). However, when I do:

ros2 param get /move_group manipulator.kinematics_solver

It is empty, while:

ros2 param get /move_group robot_description_kinematics.manipulator.kinematics_solver

Shows the correct KDL kinematics solver. Shouldn't the /move_group manipulator.kinematics_solver also be loaded correctly?

Am I missing something? Please help! Thanks.

### KUKA robot OS

iiQKA

### KUKA robot OS version

iiQKA.OS 1.2

### KUKA external interface version

1.0.3

### Affected robot model(s)

lbr iisy 3 r760

### Setup

```bash
N/A.
```

### Reproduction steps

```bash
N/A.
```

prasuchit, 2024-05-26:
**UPDATE:** By following your example code from [here](https://github.com/kroshu/kuka_drivers/blob/master/examples/iiqka_moveit_example/src/moveit_collision_avoidance_example.cpp), I realized that [this](https://github.com/kroshu/kuka_drivers/blob/e024a00b97783126d0cd61e1a98b7db343d4fa30/examples/iiqka_moveit_example/src/moveit_collision_avoidance_example.cpp#L49) line is the only difference essentially.

So I tried sending Isometry3d pose to setPoseTarget and a const std::vector<double> to setJointValueTarget and they both work.

However, if I change the quaternion to be anything other than  Eigen::Quaterniond::Identity(), like for example Eigen::Quaterniond(0, 1, 0, 0) like you have used [here](https://github.com/kroshu/kuka_drivers/blob/e024a00b97783126d0cd61e1a98b7db343d4fa30/examples/iiqka_moveit_example/src/moveit_depalletizing_example.cpp#L55C53-L55C83), the plan keeps failing.

And I still see: [WARN] [1716557929.990758924] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!

Which does not appear when I run any of your example scripts.

One last question is, in ROS1 Python, I used to be able to send a pose goal to setJointValueTarget, how can I do that here?

Hoping to hear from you soon. Thanks!

Svastits, 2024-05-27:
Hi @prasuchit, 

I have run your code, and I can see the following warning in the launch file terminal:
```[moveit_kinematic_constraints.kinematic_constraints]: Orientation constraint for link 'tool0' is probably incorrect: 0.387510, 0.996789, -0.019382, 0.067334. Assuming identity instead```
The quaternion you specify has to be a unit quaternion (I think with quite high precision), therefore it tries to plan for identity, which I assume is not a valid position. `Eigen::Quaterniond(0, 1, 0, 0)` is certainly a valid quaternion, did you check that the pose you defined for it is indeed reachable?

The warning about the kinematics plugin is also present for our example, but I have not seen any effect of this, planning works as it should.  I will look at the code when I have some time to see whether it should have some effect.

What do you mean by setting a pose using `setJointValueTarget`? 
You can either set a cartesian position using `setPoseTarget` or a joint goal using `setJointValueTarget`.

prasuchit, 2024-05-27:
@Svastits, thanks for the response. Yes, the quaternion and pose I'm sending is valid. The way I know that for sure is that I created this home pose within the SRDF (similar to the null_pos): 0.042, -1.343, 1.436, 0.03927, 1.343, -0.042.

I can use the Rviz GUI to send the robot to this pose easily. After sending it there, I print out the move_group's current pose and this is the pose that is printed.

I just checked and when I set the same pose through:

    Eigen::Isometry3d(Eigen::Translation3d(0.387445, -0.022468, 0.320447) * Eigen::Quaterniond(0.038769, 0.996789, -0.019433, 0.067307));
    
Instead of using geometry_msgs::msg::Pose msg, it works perfectly... I don't know what is different! :/

Anyway, thank you for the help. I do have a connectivity problem with the physical robot. But I can start a new issue for that.

closed 2024-05-27 by prasuchit