kroshu/kuka_drivers #168: Optimize driver threading strategy

Svastits, 2024-05-30:
On client machines with less CPU cores/available threads, the lifecycle transitions of the `robot_manager` nodes could fail due to the lack of resources.
This problem should be investigated and optimized if possible

prasuchit, 2024-10-09:
@Svastits I just moved to Jazzy and I still needed the RT kernel to connect with the robot since it wasn't working with the generic kernel. Any update on this optimization issue? Please let me know.

Svastits, 2024-10-10:
could you share some details about your setup (hardware and software), where it was not working?
We tried it with low-performance hardware and did not find any issues... 

Did you enable the configuration of priorities ([see wiki](https://github.com/kroshu/kuka_drivers/wiki/5_Realtime#configuration)) on your machine when this was not working? That is necessary even without the RT patch

prasuchit, 2024-10-10:
Hi @Svastits ,

The computer I'm using is a Dell Precision Tower with a 12GB Nvidia RTX A2000 GPU and Nvidia 560.35 driver and a CUDA 12.6 installed. It has 64GB RAM if I remember correctly. Without an RT kernel even the configure command wouldn't work. So I would say it definietly depends on the kernel a lot. Thanks for looking into this. :)