kroshu/kuka_drivers #177: Feature Request: Publish WrenchStamped Message for Force Torque Information

ekampourakis, 2024-08-05:
## Description

I would like to request the addition of a feature to the KUKA ROS2 driver to publish the force torque information of the robot arm as a `geometry_msgs/msg/WrenchStamped` [message.](https://docs.ros.org/en/humble/p/geometry_msgs/interfaces/msg/WrenchStamped.html) This feature would be similar to the common practice (e.g. implementation in this [driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/d73f7f7b4d1c978780cb19648d505678317b3009/ur_robot_driver/doc/controllers.md?plain=1#L57)), where the controller publishes the wrench measured from the robot at its TCP or other configurable frame on a topic  (e.g. `/wrench`).

## Motivation

Currently, the driver publishes torque data (effort) within the `JointState` message. While this information is useful, it does not suffice for applications that require force torque data already transformed into a specified frame or TCP. By providing this data in the form of a `WrenchStamped` message, we can enable a variety of advanced applications, such as:

1. **Probing Surfaces:** The transformed force torque data allows precise control and sensing when interacting with surfaces, enhancing the robot's capabilities in tasks such as automatic calibration.

2. **Advanced Control Schemes:** Certain control schemes, including force control and impedance control, rely on accurate force torque information relative to a specific frame. Providing this data directly in the desired frame simplifies the implementation and improves control performance.

3. **Calibration Capabilities:** Similar to other implementations, the ability to calibrate the wrench to 0 would allow for more accurate and adaptable force measurements, crucial for dynamic environments and changing tasks.

## Implementation Suggestion

- **New Topic:** Publish a `geometry_msgs/msg/WrenchStamped` message on a new topic, such as `/ns/wrench`, to provide soft real-time force torque data at the TCP or another specified frame.
- **Calibration Function:** Consider implementing a taring function to allow users to reset the wrench measurements as needed.

## Additional Information

- The `WrenchStamped` message provides a standardized format that is already widely used in ROS-based systems, ensuring compatibility and ease of integration.
- By leveraging internal data already available within the high-speed communication system of iiQKA, this feature can be implemented with minimal overhead while providing significant benefits to users.

Thank you for considering this request. I believe it will add significant value to the KUKA driver and benefit the broader robotics community.

Svastits, 2024-08-05:
Thanks for the detailed feature request. 
It will not be possible in the near future to extend the server side of the driver, so we will consider our options, how we could calculate the wrench on the ROS2 side

ekampourakis, 2024-08-06:
Thank you for considering and exploring options for calculating the wrench on the ROS2 side. I appreciate your efforts to find an alternative solution given the constraints on extending the server side.

Please tag me in any follow-ups or discussions related to this request, as I’m eager to stay updated and assist where possible.

Thanks again for your attention and collaboration!

ekampourakis, 2024-09-30:
Just wanted to follow up and see if there have been any updates on this. It would be really helpful for our project

Svastits, 2024-10-17:
Calculating wrench on the ROS2 side was deemed too complex and too much of an effort, now we are investigating the possibilities of adding it to the server side