kroshu/kuka_drivers #180: Adding a linear axis

terkaa, 2024-08-17:
Hi,

   Our KR10R1100 is installed onto a KL100 linear axis. I would like to add control of the linear axis to the RVIZ. I have managed to include the meshes and links to the URDF, but currently I have a fixed joint between the KL100 base and slide. Is it possible control for the linear axis and how to proceed?

Tero

terkaa, 2024-09-17:
Anyone?

Svastits, 2024-09-18:
Sorry, this question went under the radar...

We will not be able to update the driver in the near future to support external axis, but I think we can provide directions and some support for you. (We might be able to create a solution on a side-branch, we will update you of the approximate timeline if it is indeed possible)
A few things have to be modified to be able to add external axis support:
- **RSI context** (.rsix file) should be extended with AxisCorrExt block and first input connected to Ethernet (current version can be seen in the picture, this should remain the same and an additional block is to be added)
[image]
- **ros_rsi_ethernet.xml** should be extended with a new receive element
```<ELEMENT TAG="EK.E1" TYPE="DOUBLE" INDX="8" HOLDON="1" />```
- a new command interface should be added to the kss hardware interface
- xml parsing should be extended to copy the value of this interface to the communication xml
- lastly, a controller is needed to interact with this command interface - the joint trajectory controller could be used for this (maybe another instance is needed, but modifying the current configuration might also work)