kroshu/kuka_drivers #190: [BUG] - Compilation Issue

ekampourakis, 2024-10-15:
### Description

When trying to build the humble branch of the kuka_drivers, the packages kuka_iiqka_eac_driver and kuka_external_control_sdk_examples fail to build and complain about mismatching function arguments from the SDK.

### KUKA robot OS

iiQKA

### KUKA robot OS version

KUKA Linux OS 2.0.14

### KUKA external interface version

ExternalControl.API 1.0.3

### Affected robot model(s)

iisy R1300

### Version or commit hash of the driver

humble

### Setup

```bash
Bug during compilation
```

### Reproduction steps

```bash
colcon build --packages-select kuka_iiqka_eac_driver
```

### Logs

```bash
Starting >>> kuka_iiqka_eac_driver
--- stderr: kuka_iiqka_eac_driver                             
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp: In member function ‘virtual hardware_interface::return_type kuka_eac::KukaEACHardwareInterface::read(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:240:42: error: request for member ‘begin’ in ‘(& req_message)->kuka::external::control::BaseMotionState::GetMeasuredPositions()’, which is of pointer type ‘const std::vector<double>*’ (maybe you meant to use ‘->’ ?)
  240 |       req_message.GetMeasuredPositions().begin(), req_message.GetMeasuredPositions().end(),
      |                                          ^~~~~
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:240:86: error: request for member ‘end’ in ‘(& req_message)->kuka::external::control::BaseMotionState::GetMeasuredPositions()’, which is of pointer type ‘const std::vector<double>*’ (maybe you meant to use ‘->’ ?)
  240 |       req_message.GetMeasuredPositions().begin(), req_message.GetMeasuredPositions().end(),
      |                                                                                      ^~~
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:243:40: error: request for member ‘begin’ in ‘(& req_message)->kuka::external::control::BaseMotionState::GetMeasuredTorques()’, which is of pointer type ‘const std::vector<double>*’ (maybe you meant to use ‘->’ ?)
  243 |       req_message.GetMeasuredTorques().begin(), req_message.GetMeasuredTorques().end(),
      |                                        ^~~~~
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:243:82: error: request for member ‘end’ in ‘(& req_message)->kuka::external::control::BaseMotionState::GetMeasuredTorques()’, which is of pointer type ‘const std::vector<double>*’ (maybe you meant to use ‘->’ ?)
  243 |       req_message.GetMeasuredTorques().begin(), req_message.GetMeasuredTorques().end(),
      |                                                                                  ^~~
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp: In member function ‘virtual hardware_interface::return_type kuka_eac::KukaEACHardwareInterface::write(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:269:56: error: no matching function for call to ‘kuka::external::control::BaseControlSignal::AddJointPositionValues(std::vector<double>::iterator, std::vector<double>::iterator)’
  269 |   robot_ptr_->GetControlSignal().AddJointPositionValues(
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  270 |     hw_position_commands_.begin(), hw_position_commands_.end());
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ekampourakis/.local/include/kuka/external-control-sdk/common/irobot.h:23,
                 from /home/ekampourakis/.local/include/kuka/external-control-sdk/iiqka/sdk.h:18,
                 from /home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/include/kuka_iiqka_eac_driver/event_observer.hpp:21,
                 from /home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:25:
/home/ekampourakis/.local/include/kuka/external-control-sdk/common/message_builder.h:58:8: note: candidate: ‘void kuka::external::control::BaseControlSignal::AddJointPositionValues(std::vector<double>&)’
   58 |   void AddJointPositionValues(std::vector<double>& joint_position_values) {
      |        ^~~~~~~~~~~~~~~~~~~~~~
/home/ekampourakis/.local/include/kuka/external-control-sdk/common/message_builder.h:58:8: note:   candidate expects 1 argument, 2 provided
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:271:49: error: no matching function for call to ‘kuka::external::control::BaseControlSignal::AddTorqueValues(std::vector<double>::iterator, std::vector<double>::iterator)’
  271 |   robot_ptr_->GetControlSignal().AddTorqueValues(
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  272 |     hw_torque_commands_.begin(), hw_torque_commands_.end());
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ekampourakis/.local/include/kuka/external-control-sdk/common/irobot.h:23,
                 from /home/ekampourakis/.local/include/kuka/external-control-sdk/iiqka/sdk.h:18,
                 from /home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/include/kuka_iiqka_eac_driver/event_observer.hpp:21,
                 from /home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:25:
/home/ekampourakis/.local/include/kuka/external-control-sdk/common/message_builder.h:67:8: note: candidate: ‘void kuka::external::control::BaseControlSignal::AddTorqueValues(std::vector<double>&)’
   67 |   void AddTorqueValues(std::vector<double>& joint_torque_values) {
      |        ^~~~~~~~~~~~~~~
/home/ekampourakis/.local/include/kuka/external-control-sdk/common/message_builder.h:67:8: note:   candidate expects 1 argument, 2 provided
/home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:273:62: error: no matching function for call to ‘kuka::external::control::BaseControlSignal::AddStiffnessAndDampingValues(std::vector<double>::iterator, std::vector<double>::iterator, std::vector<double>::iterator, std::vector<double>::iterator)’
  273 |   robot_ptr_->GetControlSignal().AddStiffnessAndDampingValues(
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  274 |     hw_stiffness_commands_.begin(), hw_stiffness_commands_.end(), hw_damping_commands_.begin(),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  275 |     hw_damping_commands_.end());
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~                               
In file included from /home/ekampourakis/.local/include/kuka/external-control-sdk/common/irobot.h:23,
                 from /home/ekampourakis/.local/include/kuka/external-control-sdk/iiqka/sdk.h:18,
                 from /home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/include/kuka_iiqka_eac_driver/event_observer.hpp:21,
                 from /home/ekampourakis/kuka_ws/src/kuka_drivers/kuka_iiqka_eac_driver/src/hardware_interface.cpp:25:
/home/ekampourakis/.local/include/kuka/external-control-sdk/common/message_builder.h:85:8: note: candidate: ‘void kuka::external::control::BaseControlSignal::AddStiffnessAndDampingValues(std::vector<double>&, std::vector<double>&)’
   85 |   void AddStiffnessAndDampingValues(std::vector<double>& stiffness_values,
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ekampourakis/.local/include/kuka/external-control-sdk/common/message_builder.h:85:8: note:   candidate expects 2 arguments, 4 provided
gmake[2]: *** [CMakeFiles/kuka_iiqka_eac_driver.dir/build.make:76: CMakeFiles/kuka_iiqka_eac_driver.dir/src/hardware_interface.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/kuka_iiqka_eac_driver.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< kuka_iiqka_eac_driver [2.05s, exited with code 2]

Summary: 0 packages finished [2.23s]
  1 package failed: kuka_iiqka_eac_driver
  1 package had stderr output: kuka_iiqka_eac_driver
```

Svastits, 2024-10-15:
Hi, please update your external-control-sdk repo, as that seems to be outdated.
On [master](https://github.com/kroshu/kuka-external-control-sdk/blob/26026bfa0379b5170d0694c7a51071dffc61e7ae/kuka_external_control_sdk/common/include/kuka/external-control-sdk/common/message_builder.h#L29C1-L30C1), the GetMeasuredPositions function is not a pointer, this is what causes your issue

ekampourakis, 2024-10-15:
Thanks for the prompt reply, switching to the master branch fixed the compilation issue with the driver but the SDK examples continue to fail. 

Consider updating the installation instructions to point to the proper branch.

I am closing this issue.

Thanks again

closed 2024-10-15 by ekampourakis