kroshu/kuka_drivers #191: Question

Ghassan-Nafash, 2024-10-16:
### Description

I have read the repository description. I am wondering if the KSS (v8.3.43) software with RSI interface (v3.2.0) supports a "velocity control" mode or "Cartesian velocity control".  As i have seen, it does not. But could be implemented or does a newer version of KSS can support these two control configurations ("velocity control" mode or "Cartesian velocity control")

### KUKA robot OS

KSS

### KUKA robot OS version

KSS 8.3.43

### KUKA external interface version

RSI 3.2.0

### Affected robot model(s)

KR150 R3300 k prime

### Version or commit hash of the driver

master branch

### Setup

```bash
It is just the question if the KSS 8.3.43 with the Interface RSI 3.2.0 can support "velocity control" mode or "Cartesian velocity control" modes.
```

### Reproduction steps

```bash
It is just the question if the KSS 8.3.43 with the Interface RSI 3.2.0 can support "velocity control" mode or "Cartesian velocity control" modes.
```

Ghassan-Nafash, 2024-10-16:
sorry for not setting the correct label for the issue. It is not a bug.

Svastits, 2024-10-17:
Hi, RSI only supports position control (both joint and cartesian, but our driver currently does not have cartesian position control).
Newer RSI versions also do not add the velocity feature, however I think it could be worked around using the RSI context. 

In the end, one has to provide position inputs to the control loop using RSI, but the RSI context could be modified to accept velocity inputs and calculate the positions based on these.
Unfortunately we do not have the resources to add this feature to the driver soon, but we can support you, if you are open to implementing it yourself :)