kroshu/kuka_drivers #198: Issues Launching Dual Arm Configuration with KUKA Drivers: Namespace and Controller Issues

ekampourakis, 2024-10-31:
We are experiencing significant difficulties in setting up a dual arm configuration using the provided KUKA launch files and drivers. We have both robotic arms integrated within a single URDF with the appropriate hardware systems configured, as we have additional links in the kinematic chain that are fixed and currently hardcoded.

What We Have Tried:

* Added namespaces (/right, /left) to the drivers and all relevant configurations.
* Experimented with different combinations and permutations of the configurations, adjusting both driver and controller settings.
* Used the provided launch files from the KUKA repository, as well as attempting to adapt the dual testing launch file by configuring it with our specific robot IP addresses.
    
Despite these attempts, we have encountered persistent issues such as controllers failing to load, errors relating to missing type definitions, and namespace conflicts or the control node claiming all hardware resources from the whole URDF instead of only the namespaced one.

I will add our launch file for reference, as well as the trace logs from the command line, detailing the exact commands we used and the resulting error messages. Any insights into the proper setup or configuration adjustments that could help resolve these issues would be greatly appreciated.

Launch File:
```
#!/usr/bin/env python3

import os
from turtle import right
import xacro

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import OpaqueFunction
from launch_ros.actions import Node, LifecycleNode
from launch.substitutions import PathJoinSubstitution
from launch_ros.substitutions import FindPackageShare
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from moveit_configs_utils import MoveItConfigsBuilder

def launch_setup(context, *args, **kwargs):
    # Hardcoded arguments
    left_controller_ip = "192.168.1.200"
    right_controller_ip = "192.168.1.201"
    client_ip = "192.168.1.103"

    # Config file paths
    controllers_config = os.path.join(get_package_share_directory("project_cell"), "config","controllers.yaml")
    right_jtc_config = os.path.join(get_package_share_directory("project_cell"), "config", "right_joint_trajectory_controller_config.yaml")
    right_jic_config = os.path.join(get_package_share_directory("project_cell"), "config", "right_joint_impedance_controller_config.yaml")
    right_ec_config = os.path.join(get_package_share_directory("project_cell"), "config", "right_effort_controller_config.yaml")
    right_driver_config = os.path.join(get_package_share_directory("project_cell"), "config", "right_driver_config.yaml")
    left_jtc_config = os.path.join(get_package_share_directory("project_cell"), "config", "left_joint_trajectory_controller_config.yaml")
    left_jic_config = os.path.join(get_package_share_directory("project_cell"), "config", "left_joint_impedance_controller_config.yaml")
    left_ec_config = os.path.join(get_package_share_directory("project_cell"), "config", "left_effort_controller_config.yaml")
    left_driver_config = os.path.join(get_package_share_directory("project_cell"), "config", "left_driver_config.yaml")

    # Get URDF via xacro
    robot_urdf_xacro = os.path.join(
        get_package_share_directory('project_cell'),
        'urdf',
        'project_cell.urdf.xacro'
    )
    robot_description_xml = xacro.process_file(
        robot_urdf_xacro,
        mappings={
            "use_real": "true",
            "client_ip": client_ip,
            "left_controller_ip": left_controller_ip,
            "right_controller_ip": right_controller_ip,
        }
    ).toxml()
    robot_description = {'robot_description': robot_description_xml}

    right_control_node = Node(
        namespace="/right",
        package="kuka_drivers_core",
        executable="control_node",
        parameters=[
[... 652 lines cut ...]
[control_node-1] [INFO] [1730372773.335221654] [resource_manager]: Successful 'activate' of hardware 'right_system'
[control_node-4] [ERROR] [1730372773.341613522] [KukaEACHardwareInterface]: External control stopped by an error
[control_node-4] [ERROR] [1730372773.341622557] [KukaEACHardwareInterface]: Execution failed
[control_node-1] [ERROR] [1730372773.341705812] [KukaEACHardwareInterface]: External control stopped by an error
[control_node-1] [ERROR] [1730372773.341712248] [KukaEACHardwareInterface]: Execution failed
[control_node-1] [ERROR] [1730372773.341723186] [KukaEACHardwareInterface]: External control stopped by an error
[control_node-1] [ERROR] [1730372773.341727258] [KukaEACHardwareInterface]: Could not create UDP communication, first response did not arrive (in time)
[control_node-4] [ERROR] [1730372773.341777135] [KukaEACHardwareInterface]: External control stopped by an error
[control_node-4] [ERROR] [1730372773.341784906] [KukaEACHardwareInterface]: Could not create UDP communication, first response did not arrive (in time)
[control_node-1] [INFO] [1730372773.348803059] [KukaEACHardwareInterface]: Sending stop signal
[robot_manager_node-2] [INFO] [1730372773.389395590] [right.robot_manager]: Successfully activated controllers
[control_node-4] [INFO] [1730372781.538240594] [resource_manager]: 'configure' hardware 'left_system' 
[control_node-4] [INFO] [1730372781.538254997] [left.control_mode_handler]: Control mode changed to 1
[control_node-4] [INFO] [1730372781.539522831] [KukaEACHardwareInterface]: Set QoS profile with 2 consequent and 1 packet losses allowed in 5000 milliseconds
[control_node-4] [INFO] [1730372781.539532064] [resource_manager]: Successful 'configure' of hardware 'left_system'
[control_node-4] [WARN] [1730372781.539536891] [resource_manager]: (hardware 'left_system'): 'state/server_state' state interface already in available list. This can happen due to multiple calls to 'configure'
[control_node-4] [WARN] [1730372781.539541307] [resource_manager]: (hardware 'left_system'): 'runtime_config/control_mode' command interface already in available list. This can happen due to multiple calls to 'configure'
[control_node-4] [INFO] [1730372781.539745155] [left.controller_manager]: Switch controller timeout is set to 0, using default 1s!
[robot_manager_node-5] [INFO] [1730372781.581901996] [left.robot_manager]: Activated control mode handler and event broadcaster

[control_node-4] [INFO] [1730372791.551701223] [resource_manager]: 'activate' hardware 'left_system' 
[control_node-4] [ERROR] [1730372794.608321088] [KukaEACHardwareInterface]: Starting external control failed, error message: Could not start external control
[control_node-4] [INFO] [1730372794.608346861] [resource_manager]: Failed to 'activate' hardware 'left_system'
[robot_manager_node-5] [ERROR] [1730372794.608454805] [left.robot_manager]: Could not activate hardware interface
```

ekampourakis, 2024-11-11:
We managed to load and get joint states from both robots at the same time by changing the `startup.launch.py` file and removing the namespace from the `controller_spawner` function but still cannot move both arms from MoveIt.

Both arms activate using the lifecycle command `ros2 lifecycle set /right/robot_manager activate` and `ros2 lifecycle set /left/robot_manager activate` and publish the appropriate joint states. But only the last activated control node is accepting and executing commands.

The lastly activated control node stops the other external control channel:

```[control_node-3] [INFO] [1731323345.762343791] [resource_manager]: 'activate' hardware 'right_lbr_iisy11_r1300' 
[control_node-3] [INFO] [1731323348.965581671] [KukaEACHardwareInterface]: External control session started successfully
[control_node-3] [INFO] [1731323348.965614517] [resource_manager]: Successful 'activate' of hardware 'right_lbr_iisy11_r1300'
[control_node-3] [INFO] [1731323348.965749008] [KukaEACHardwareInterface]: External control is active
[control_node-12] [ERROR] [1731323348.970883392] [KukaEACHardwareInterface]: External control stopped by an error
[control_node-12] [ERROR] [1731323348.970893623] [KukaEACHardwareInterface]: Execution failed
[control_node-12] [ERROR] [1731323348.970909147] [KukaEACHardwareInterface]: External control stopped by an error
[control_node-12] [ERROR] [1731323348.970910779] [KukaEACHardwareInterface]: Joint limits violated
[robot_manager_node-4] [INFO] [1731323348.993286244] [right.robot_manager]: Successfully activated controllers
```

Additionally this is also the case for the provided dual arm example launch file which doesn't include MoveIt.

An update on this issue could be very useful on moving on with the dual arm setup.

Svastits, 2024-11-12:
Thanks for reporting this issue, we previously did not have problems with the dual arm setup without moveit, we could launch the example, activate both drivers and move both robot with the `rqt_joint_trajectory_controller`. 
The moveit configuration files are indeed not set up to support a dual arm scenario, we will start working on that soon and will also try to investigate what could have gone wrong without moveit

Svastits, 2024-12-20:
Hi, could you find the reason for the crashes?

I started to rework the example, as the current version indeed has namespace issues. The [feature/multi_robot](https://github.com/kroshu/kuka_drivers/tree/feature/multi_robot) branch currently works for me with `use_fake_hardware` set to true and I can move the robot with directly publishing to the joint trajectory controller's topic:

`ros2 topic pub test1/joint_trajectory_controller/joint_trajectory trajectory_msgs/msg/JointTrajectory "{joint_names: ['test1_joint_1', 'test1_joint_2', 'test1_joint_3', 'test1_joint_4', 'test1_joint_5', 'test1_joint_6'], points: [{positions: [-0.2, -1.57, -0.2, -0.2, -0.2, -0.2], velocities: [], accelerations: [], effort: [], time_from_start: {sec: 4, nanosec: 0}}]}"`

I did not have the change to test it on real robots yet, is it possible, that using the same port for both causes the crash?