kroshu/kuka_drivers #201: Uncertainty on which network interface to use with the FRI drivers

thettasch, 2024-12-05:
I have been attempting to set up a Kuka iiwa for use with ROS2, in collaboration with @gavanderhoorn . We managed to get the driver to connect using the X66 network port, after unsuccessfully attempting to connect over the KONI port. It is our understanding that the KONI port should be used for real-time applications such as using the FRI driver (the Kuka FRI manual explicitly recommends use of the KONI and a point-to-point connection). This prompted the following questions:

- Is the driver intended to use the X66 network interface?
	- If yes, is reliable control of the robot possible over this interface?
- Is there potentially some error in the networking configuration of the robot controller, preventing use of the KONI port?
	- If yes, what steps would you recommend to solve the problem?

### Setup details: Robot
- Kuka LBR iiwa 14 R820
- OS: Sunrise.OS 1.16
- Configured using Kuka Sunrise Workbench 1.16
- X66 IP set to 172.31.1.147
- KONI IP set to 192.170.10.2  (192.168.10.2 also tested with same results)
  - pings to this IP return duplicate responses

### Setup details: ROS Client PC
- OS: Ubuntu 22.04.5 LTS
- Kernel: 5.15.0-1073-realtime
- ROS2 humble running inside a docker container (with `--net=host`)
- No active firewalls
- robot_drivers commit 2c84eda (humble branch)
- All configurations default
- `ros2 launch kuka_sunrise_fri_driver startup.launch.py controller_ip:=172.31.1.147 client_ip:=172.31.1.145`

gavanderhoorn, 2024-12-06:
@thettasch: thanks for the issue.

Should this have been opened as a Discussion item instead though?

thettasch, 2024-12-06:
Indeed it probably should have been.

Alternatively, perhaps I should have framed the issue as:
"_The documentation does not cover how to properly configure the robot controller, nor how to set up the networking connection between the controller and the client._"

gavanderhoorn, 2024-12-12:
Would you perhaps have any guidance @Svastits?

Svastits, 2024-12-19:
Hi, 

I am quite sure that we have always used the driver with the KLI (X66) and we did not have any stability issues, although I think we never really went to the range of 1-2ms cycle time, rather 5-10ms.

I will try to make the driver work with KONI sometime next year and update the doc in every case.

mycode-CT, 2025-02-02:
been facing the same issue, as an alternative, KONI with 1ms cycle rate appears to be supported at https://github.com/lbr-stack/lbr_fri_ros2_stack , worked for me

Svastits, 2025-03-25:
Sorry for the late response...

I could test the issue a little more in detail and I can confirm, that KONI is not working, as the TCP connection initialization fails. I could not find the reason for this yet, but the driver I started a couple of ours ago with KLI and 1ms cycle time is still running.

So in my opinion the following should be done:
- extend the wiki with the fact, that currently only KLI is working, but it seems to run stable enough even with the lowest cycle time
- investigate why the TCP connection is not available on KONI

I would close this issue, when the first is done and create a new one for the second point

Svastits, 2025-03-26:
update: I have stopped the "endurance run" after 24 hours

Svastits, 2025-03-26:
I have updated the wiki and created a new issue for supporting KONI (#221)

closed 2025-03-26 by Svastits