kroshu/kuka_drivers #210: [BUG] - Could not start FRI session with FRI 2.5

RLi43, 2025-02-10:
### Description

FRI session can not be established.

From my understanding, in hardware activation, the hardware component `KukaFRIHardwareInterface` send StartFRI request and wait for the result. On the robot application side, it runs `FRIManager.this._FRISession.await(10, TimeUnit.SECONDS);`. The case is, during this activation, the hardware is not updated and FRI session didn't get reply from our PC, so fails to start FRI session and returns `Command errored`. 

I added debug output in `KukaFRIHardwareInterface::read` and `KukaFRIHardwareInterface::write` as I suspected that when setting the hardware component state, the hardware update loop is freeze (There's a resource lock in [set_component_ state](https://github.com/ros-controls/ros2_control/blob/dff147424875df0215734970e5527eeafd6e5d72/hardware_interface/src/resource_manager.cpp#L1228C47-L1228C62) and [read](https://github.com/ros-controls/ros2_control/blob/dff147424875df0215734970e5527eeafd6e5d72/hardware_interface/src/resource_manager.cpp#L1252) and write). 

### KUKA robot OS

Sunrise

### KUKA robot OS version

2.5

### KUKA external interface version

/

### Affected robot model(s)

LBR

### Setup

```bash
using `humble` branch
robot application code modified accordingly to fit FRI 2.5
```

### Reproduction steps

```bash
ros2 launch kuka_sunrise_fri_driver startup.launch.py controller_ip:=172.31.1.147

# Another Terminal
ros2 lifecycle set robot_manager configure # Successful
ros2 lifecycle set robot_manager activate # Failed
```

### Logs

```bash
1739184952.4846172 [INFO] [launch]: All log files can be found below /home/jiangfan/.ros/log/2025-02-10-11-55-52-483831-KneeRSDesktop-78407
1739184952.4847772 [INFO] [launch]: Default logging verbosity is set to INFO
1739184952.9592102 [INFO] [control_node-1]: process started with pid [78420]
1739184952.9661102 [INFO] [robot_manager_node-2]: process started with pid [78422]
1739184952.9725249 [INFO] [robot_state_publisher-3]: process started with pid [78424]
1739184952.9730127 [INFO] [spawner-4]: process started with pid [78426]
1739184952.9732594 [INFO] [spawner-5]: process started with pid [78428]
1739184952.9734557 [INFO] [spawner-6]: process started with pid [78430]
1739184952.9736419 [INFO] [spawner-7]: process started with pid [78432]
1739184952.9738479 [INFO] [spawner-8]: process started with pid [78434]
1739184952.9740369 [INFO] [spawner-9]: process started with pid [78436]
1739184952.9742188 [INFO] [spawner-10]: process started with pid [78459]
1739184952.9744151 [INFO] [spawner-11]: process started with pid [78466]
1739184952.9745972 [INFO] [spawner-12]: process started with pid [78469]
1739184952.9867909 [robot_state_publisher-3] [INFO] [1739184952.985925372] [robot_state_publisher]: got segment base
1739184952.9869583 [robot_state_publisher-3] [INFO] [1739184952.986088846] [robot_state_publisher]: got segment base_link
1739184952.9870567 [robot_state_publisher-3] [INFO] [1739184952.986109353] [robot_state_publisher]: got segment flange
1739184952.9871485 [robot_state_publisher-3] [INFO] [1739184952.986120964] [robot_state_publisher]: got segment link_1
1739184952.9872341 [robot_state_publisher-3] [INFO] [1739184952.986129752] [robot_state_publisher]: got segment link_2
1739184952.9873145 [robot_state_publisher-3] [INFO] [1739184952.986140542] [robot_state_publisher]: got segment link_3
1739184952.9873917 [robot_state_publisher-3] [INFO] [1739184952.986154785] [robot_state_publisher]: got segment link_4
1739184952.9874697 [robot_state_publisher-3] [INFO] [1739184952.986167490] [robot_state_publisher]: got segment link_5
1739184952.9884133 [robot_state_publisher-3] [INFO] [1739184952.986176492] [robot_state_publisher]: got segment link_6
1739184952.9885201 [robot_state_publisher-3] [INFO] [1739184952.986185227] [robot_state_publisher]: got segment link_7
1739184952.9886062 [robot_state_publisher-3] [INFO] [1739184952.986193929] [robot_state_publisher]: got segment tool0
1739184952.9886889 [robot_state_publisher-3] [INFO] [1739184952.986202632] [robot_state_publisher]: got segment world
[... 96 lines cut ...]
1739184997.0859606 [control_node-1] Error: failed while trying to receive monitoring message!
1739184997.0861197 [control_node-1] [INFO] [1739184997.085614124] [KukaFRIHardwareInterface]: Debug: Hardware Write
1739184997.0861824 [control_node-1] [INFO] [1739184997.085630562] [KukaFRIHardwareInterface]: Debug: Hardware Read
1739184997.0868261 [control_node-1] The connection has timed out. Timeout is 1

...

1739184999.7101405 [control_node-1] [INFO] [1739184999.708911281] [KukaFRIHardwareInterface]: Debug: Hardware Write
1739184999.7102072 [control_node-1] [INFO] [1739184999.708942328] [KukaFRIHardwareInterface]: Debug: Hardware Read
1739184999.7104418 [control_node-1] The connection has timed out. Timeout is 1
1739184999.7105045 [control_node-1] Error: failed while trying to receive monitoring message!
1739184999.7107368 [control_node-1] [INFO] [1739184999.709978516] [KukaFRIHardwareInterface]: Debug: Hardware Write
1739184999.7108171 [control_node-1] [INFO] [1739184999.709994417] [KukaFRIHardwareInterface]: Debug: Hardware Read
1739184999.7112532 [control_node-1] The connection has timed out. Timeout is 1
1739184999.7113333 [control_node-1] Error: failed while trying to receive monitoring message!
1739184999.7116015 [control_node-1] [INFO] [1739184999.711028358] [resource_manager]: 'cleanup' hardware 'lbr_iiwa14_r820' 
1739184999.7122436 [control_node-1] [INFO] [1739184999.711979913] [resource_manager]: Successful 'cleanup' of hardware 'lbr_iiwa14_r820'
1739185124.5106659 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
1739185124.7590113 [robot_state_publisher-3] [INFO] [1739185124.510430216] [rclcpp]: signal_handler(signum=2)
1739185124.7592890 [INFO] [robot_state_publisher-3]: process has finished cleanly [pid 78424]
1739185124.7594626 [INFO] [robot_manager_node-2]: process has finished cleanly [pid 78422]
1739185124.7596109 [INFO] [control_node-1]: process has finished cleanly [pid 78420]
1739185124.7599473 [robot_manager_node-2] [INFO] [1739185124.510431156] [rclcpp]: signal_handler(signum=2)
1739185124.7603576 [control_node-1] [INFO] [1739185124.510430655] [rclcpp]: signal_handler(signum=2)
```

Hydran00, 2025-02-19:
Hi @RLi43,
I'm encountering the exact same issue you reported. I am using ROS2 Humble with a KUKA LBR 14. The activation procedure attempts a service call but times out. Initially, I suspected a spinning issue (where the response is never received), but I also tried invoking the service manually from the ROS2 CLI and got the same error. Did you manage to find a workaround?

RLi43, 2025-02-19:
Hi @Hydran00, I launched a thread handling the FRI communication so that it won't be interrupted.

RLi43, 2025-02-19:
btw, the log here shows a problem in FRI connection (Fri session state didn't move to monitoring_ready), but the activation problem persists after fixing this.

Hydran00, 2025-02-19:
Thanks for your response! Then I think I'll wait for a fix from the maintainers.

I'm currently looking to implement variable impedance control by updating stiffness and damping in real time. I noticed that this repository seems to support parameterizing these values, but it's unclear to me whether they can be updated in real time or only before the control starts. Do you know if this is actually possible?

Hydran00, 2025-02-19:
Thank you! Indeed, I was using FRI with the KUKA impedance controller. What I don't understand is how it is possible to change the stiffness in real-time without stopping the FRI overlay, and if this repository provides that possibility.

Svastits, 2025-02-20:
Thanks for reporting this issue, it seems an update to `ros2_control` indeed introduced a new mutex, that blocks activation of the FRI driver. Cyclic communication is necessary for FRI after startFRI is called, but activating control blocks this communication due to the new mutex. We will try to fix as soon as possible.

Unfortunately as far I know, it is not possible to update the joint impedance attributes in realtime for FRI - the ROS2 side is prepared for this, as it is possible for iiQKA robots, but for FRI, the attributes must be specified when starting the motion overlay and cannot be modified later...

closed 2025-03-25 by Svastits