kroshu/kuka_drivers #256: [BUG] - Robot Manager Node not Configuring for EKI RSI Kuka (KR 240 R2900_2) Robot

james94, 2025-09-24:
### Description

First I made sure that I could deploy the open source **moveit_planning_example.launch.py** script with the **kuka lbr_iisy3_r760**, configure and activate the robot manager and then run the **moveit_basic_planners_example.cpp** to use **pilz_industiral_motion_planner** with **PTP, LIN** approaches and see the kuka lbr robot move successfully in ROS2 RViz simulation. Both of these programs, come from the ROS2 Kuka Drivers open source repo.

However, when I modified the launch script to leverage **kuka kr240_r2900_2**, I was able to successfully launch the kuka kr robot model in RViz, but when I tried to **configure** the robot manager, the logs from the **moveit_planning_example.launch.py** script showed an error related to """[ERROR] [1758736181.071563441] [resource_manager]: Not acceptable command interfaces combination""".  I even tried deploying the **moveit_planning_example.launch.py** script with the **kuka kr210_r2700_2** and got the similar error. I am working on trying to deploy the **kuka kr240_r2900_2** robot and running path planning using **pilz_industiral_motion_planner** with **PTP** for RViz simulation mode and then on the real kuka robot. I will provide the logs too. Can you help me resolve this problem?

Also I was able to use **pilz_industiral_motion_planner** with **PTP** in ROS2 RVIz MoveIt GUI and see the robot movement and the **kuka kr240_r2900_2** was configured in **rsi_only** mode. However, when I tried to run the **moveit_planning_example.launch.py** script with **kuka kr240_r2900_2** in RSI only mode, I was able to configure and activate the robot manager successfully, and then when I tried testing the **PTP** planning from **pilz**, I got an error related to """[move_group-6] [ERROR] [1758734173.016274143] [moveit.pilz_industrial_motion_planner.trajectory_functions]: Unable to find IK solution. [move_group-6] [ERROR] [1758734167.368889664] [moveit.pilz_industrial_motion_planner.trajectory_generator]: Failed to compute inverse kinematics for link: tool0 of goal pose""" right after receiving the moveit request, executing..., planning request received for MoveGroup action, using planning pipeline pilz_industiral_motion_planner, initialized point to point trajectory generator and generating PTP trajectory. However, this problem didn't happen with **kuka lbr_iisy3_r760** when I ran the example.

### KUKA robot OS

KSS

### KUKA robot OS version

KSS 8.6.10

### KUKA external interface version

RSI 4.1

### Affected robot model(s)

KR 240 R2900_2

### Version or commit hash of the driver

91f6c1d432decb31b7225f8217bfbdeca1ec3a85

### Setup

```bash
I am using a slightly different version of the "moveit_planning_example.launch.py" script, which I call "moveit_planning_example_open.launch.py" and a slightly different version of the "startup.launch.py" script from ROS2 Kuka Drivers, so it can work with the "kuka kr240_r2900_2" robot model. So, when I run the launch py script, I run it as follows and pass the arguments: "ros2 launch iiqka_moveit_example moveit_planning_example_open.launch.py mode:=mock driver_version:=eki_rsi robot_model:=kr240_r2900_2 robot_family:=quantec". Then I configure the robot manager, but then I see errors in the logs of the launch py script. I am using the same version of the "moveit_basic_planners_example.cpp" since that can work with either "kuka lbr_iisy3_r760" or "kuka kr240_r2900_2" and other kuka robot models and we just send pilz planning for PTP and LIN movements to our ROS2 nodes that were deployed from our launch py script.
```

### Reproduction steps

```bash
1\. Build and install the **KUKA Drivers** ROS2 Package and its dependency ROS2 packages automatically through the **upstream.repos** file.

2\. Copy the **kr240_r2900_2** URDF from master branch of the kuka robot descriptions repo and its dependencies into the kuka_quantec_support folder of the kuka_robot_descriptions repo. Then rebuild kuka_robot_descriptions ROS2 package and its dependency ROS2 packages for **KUKA Drivers** ROS2 Package.

3\. Then we can execute the following commands to launch "moveit_planning_example.launch.py" script, so we can aim to run "pilz_industiral_motion_planner" with "PTP" or "LIN" mode for "kuka kr240_r2900_2" robot model:

~~~bash
##
#   Terminal 1
##
# NOTE: the open source launch script and its dependency launch script will
# need to be modified to work with kuka kr robots alternative to kuka lbr
ros2 launch iiqka_moveit_example moveit_planning_example.launch.py mode:=mock driver_version:=eki_rsi robot_model:=kr240_r2900_2 robot_family:=quantec

##
#   Terminal 2
##
ros2 lifecycle set robot_manager configure
# NOTE: When Kuka Kr is deployed in EKI_RSI, configure fails
ros2 lifecycle set robot_manager activate
# NOTE: When Kuka Kr is deployed in RSI only, configure and activate succeed

##
#   From terminal 2
##
# NOTE: In the case we deploy Kuka Kr in RSI Only mode, we'd get to 
# the point of launching this program; In EKI RSI, we dont make it this far.
cd /path/to/ros_ws/install/iiqka_moveit_example/lib/iiqka_moveit_example
./moveit_basic_planners_example
~~~
```

### Logs
[... 188 lines cut ...]
[control_node-1] Start interfaces:
[control_node-1] [
[control_node-1]   runtime_config/control_mode
[control_node-1]   runtime_config/drive_state
[control_node-1]   runtime_config/cycle_time
[control_node-1] ]
[control_node-1] Stop interfaces:
[control_node-1] [
[control_node-1] ]
[control_node-1] Not existing:
[control_node-1] [
[control_node-1]  runtime_config/drive_state
[control_node-1]  runtime_config/cycle_time
[control_node-1] ]
[control_node-1]
[control_node-1] [ERROR] [1758736181.071579889] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[control_node-1] [WARN] [1758736181.071953781] [controller_manager]: Controller with name 'control_mode_handler' can not be deactivated since it is not active.
[control_node-1] [WARN] [1758736181.071971232] [controller_manager]: Could not deactivate controller with name 'control_mode_handler'. Check above warnings for more details. Check the state of the controllers and their required interfaces using `ros2 control list_controllers -v` CLI to get more information.
[control_node-1] [ERROR] [1758736181.071976114] [controller_manager]: Aborting, no controller is switched! (::STRICT switch)
[robot_manager_node_eki_rsi-2] [ERROR] [1758736181.071751650] [robot_manager]: Could not activate control mode handler or event broadcaster controller
[robot_manager_node_eki_rsi-2] [ERROR] [1758736181.072220104] [robot_manager]: Could not deactivate control mode handler and event broadcaster
[control_node-1] [INFO] [1758736181.072544837] [resource_manager]: 'cleanup' hardware 'kr240_r2900_2'
[control_node-1] [INFO] [1758736181.072563530] [resource_manager]: Successful 'cleanup' of hardware 'kr240_r2900_2'
[ERROR] [rviz2-10]: process has died [pid 22228, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/solideon/ros_ws/install/kuka_resources/share/kuka_resources/config/planning_6_axis.rviz --ros-args --log-level error --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_qk7ap0ep'].
```

Svastits, 2025-09-25:
@james94 I think I found the issue, we forgot to add the runtime_config/drive_state and runtime_config/cycle_time command interfaces to our mock hardware interface. We will fix this, but meanwhile if you would like to use the driver in mock mode, you can use the rsi-only version, as in mock mode there is no difference between the two.

james94, 2025-09-25:
@Svastits @pasztork Turns out my **pilz_industrial_motion_planner** with **PTP** planning algorithms was failing to solve IK solution and compute inverse kinematics because the waypoints were outside the Kuka KR robot's workspace. So, in RViz, I manually moved the robot, and used a joint states subscriber node to pull waypoints that are within the robot's workspace. So, now I construct my waypoints within the Kuka KR robot's workspace, so I avoid getting that IK solver error. 

I switched to using my C++ ROS2 Robot Path App to take in the waypoints that are now within the Kuka Robot's workspace, my Robot Path App creates the path plans and planned trajectories and then sends them as MoveIt requests to my ROS2 Robot Control App (, which was based on the ROS2 launch py script that I talked about in GitHub #255 ) and am able to see my Kuka KR 240 R2900_2 move across the waypoints in RViz simulation. 

I would imagine that the **moveit_basic_planners_example.cpp** most likely fails for Kuka KR robots compared to Kuka LBR maybe because those waypoints are outside the Kuka KR robot's workspace, but this isn't the case for Kuka LBR robot I mentioned earlier.